sphero-inc/sphero.js #22: Calibrating the sphero using set heading

jacensherman, 2015-06-30:
Hi, I'm one of the MSPs working on the Sphero pilot program. I've been doing a lot of research trying to figure out how to calibrate the Sphero without physically picking it up.

I found that on other platforms the way of doing this is to start calibration and then set heading (I believe). How might I do this with node.

I already have a custom element slider that generates a number between 0 and 360 that the user can use to calibrate the Sphero.

Thanks so much!

zankich, 2015-06-30:
@jacensherman this code will rotate the sphero in place and set the new heading.

```javascript
var heading = 90;
orb.roll(1, heading, 2, function() {
  setTimeout(function() {
    orb.setHeading(0, function() {
      orb.roll(0,0,1);
    });
  }, 300);
});
```

deadprogram, 2015-06-30:
Worth adding to the README?

jacensherman, 2015-06-30:
Thanks so much! I've got it pretty much working! The only thing that's a little confusing is that it would seem the calibration functions would allow for remote rotation. But, this way does also make sense.

dronious, 2015-10-29:
How do you set the rotation rate? This is in other SDK's but is not mentioned in this one?

olcar, 2016-01-16:
@dronious https://github.com/orbotix/sphero.js/blob/800142d15d0a3276f3df529e8306dd295e579a16/lib/devices/sphero.js#L64

olcar, 2016-01-16:
@jacensherman @deadprogram do you think there's anything worth mentioning in the README?

If not, I think we can easily close that issue.

NAllred91, 2018-02-07:
@zankich What is the purpose of the third parameter in your roll command?  I see that it represents state, but I can't seem to find any documentation that indicates what its for.  I'm also having trouble with the rotation rate, and I'm wondering if that third parameter has something to do with it?

JudithLim, 2018-06-25:
@NAllred91 The third parameter can be used to command Sphero to stop rolling (true = stop; false = don't stop)