sphero-inc/sphero.js #23: Packet._verifyChecksum - error is unhandled

Foxman13, 2015-07-13:
This happens when sending roll() commands repeatedly.

Full error stack -

ERROR: Error: Incorrect checksum, packet discarded!                                                                                                                                                                                                                  
    at Packet._verifyChecksum (/home/root/.node_app_slot/node_modules/sphero/lib/packet.js:152:24)                                                                                                                                                                   
    at Packet._parse (/home/root/.node_app_slot/node_modules/sphero/lib/packet.js:124:15)                                                                                                                                                                            
    at Packet.parse (/home/root/.node_app_slot/node_modules/sphero/lib/packet.js:85:21)                                                                                                                                                                              
    at Adaptor.<anonymous> (/home/root/.node_app_slot/node_modules/sphero/lib/sphero.js:98:34)                                                                                                                                                                       
    at Adaptor.emit (events.js:117:20)                                                                                                                                                                                                                               
    at SerialPort.emit (events.js:95:17)                                                                                                                                                                                                                             
    at Object.module.exports.raw [as parser] (/home/root/.node_app_slot/node_modules/sphero/node_modules/serialport/parsers.js:8:13)                                                                                                                                 
    at Object.SerialPort.opts.dataCallback (/home/root/.node_app_slot/node_modules/sphero/node_modules/serialport/serialport.js:181:12)                                                                                                                              
    at SerialPortFactory.SerialPort._emitData (/home/root/.node_app_slot/node_modules/sphero/node_modules/serialport/serialport.js:396:20)                                                                                                                           
    at afterRead (/home/root/.node_app_slot/node_modules/sphero/node_modules/serialport/serialport.js:372:20)

edgarsilva, 2015-07-13:
@Foxman13 this is expected behaviour to give developers the option to listen/catch the error and do something about it if they want to. Maybe we should add a flag or something on weather to just discard the packages or actually throw errors and halt execution.

In this case you can catch it with something like:

```javascript
var sphero = require("sphero"),
    orb = sphero("/dev/rfcomm0");

orb.connect(function() {
  // Sphero's connected!
  // do some cool stuff here!
  orb.on("error", function(err, data) {
    // Do something with the err or just ignore.
  });
});
```

deadprogram, 2015-12-24:
Hi, @Foxman13 just had a chance to look into this, and have implemented @edgarsilva suggestion for allowing the user to choose to  set the `emitPacketErrors` option to `true` when creating a new connection to a Sphero. The new default is `false`, so bad packets can be gracefully handled in the simplest case. TO get the errors to be emitted as in the example @edgarsilva provided, you will now need to set `emitPacketErrors`

```
var sphero = require("sphero"),
    orb = sphero("/dev/rfcomm0", {emitPacketErrors: true});

orb.connect(function() {
  // Sphero's connected!
  // do some cool stuff here!
  orb.on("error", function(err, data) {
    // Do something with the err or just ignore.
  });
});
```

I will leave this issue open until the new code is merged into master.

deadprogram, 2015-12-24:
This is now part of the `master` branch, and docs have been added to the README here: https://github.com/orbotix/sphero.js#error-handling

I will close this issue, please reopen if further work is needed. Thank you!

closed 2015-12-24 by deadprogram