sphero-inc/sphero.js #25: Added connection via bluetooth.

vincent-leonardo, 2015-08-24:
Hi, I've encountered a difficulty connecting through serial-port library (refer to hybridgroup/cylon#310). I've added a new adaptor to handle bluetooth connection via MACAddress and modified the loader.js to automatically handle port input with bluetooth MACAddress. You can check out the codes at my fork (https://github.com/vincent-leonardo/sphero.js/commit/f2658f612b232032d6d2d49e5e500e7cd0e354da). If you don't mind, I could create a pull request for it.

deadprogram, 2015-08-25:
I'll check it out, thank you @vincent-leonardo

olcar, 2016-01-16:
Hi @vincent-leonardo, can you tell us a bit more about it? How can we test this?

Are you running on Windows 10?

vincent-leonardo, 2016-01-17:
Hi @olcar, I'm really sorry I can't remember exactly what was the problem as it has been few months back. But yes I'm running Windows 10.

The rough scenario is as follow:
If I'm not mistaken it was the issue with establishing connection with the COM port. It's either a "Semaphore Time Out (Error Code: 121" or "Element Not Found (Error Code: 1168)".

So I've incoperated bluetooth-serialport library to handle the connection. With that library I could establish connection just based on MAC Address of the sphero/device (obtained through scanning). The adaptor itself will then select the appropriate COM port to establish the connection.

Dealerpriest, 2017-01-09:
Hello @vincent-leonardo. Is your alternative bluetooth implementation still valid? I actually stumble upon the problem you describe here https://github.com/hybridgroup/cylon/issues/310#issue-102175336.
I succeed in connecting approximately every 3rd try. The other times it throws
```
events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: Opening COM79: Unknown error code 121
    at Error (native)
```

vincent-leonardo, 2017-01-10:
I'm sorry, it's been awhile since I've used this library. Now I don't have a Sphero to work with, so I couldn't test it for you. And my use case last time didn't require me reconnecting devices multiple times, hence I'm not sure whether such issue persists.

Dealerpriest, 2017-01-10:
I actually ended up copy/pasting your adapter code and the edits in
loader.js for a quick and dirty test. Unfortunately the problem persists,
with the connection failing ever so often. Thanks anyway for providing some
alternative adapter code!