sphero-inc/sphero.js #26: Support for developing on BB8

mcgrews3, 2015-09-05:
Hello,

Just got a new BB8 and am a node.js developer.  Noticed there are resources and info about programming Ollie + Sphero but nothing on BB8.  

* Is development support for BB8?   
* If so - what SDKs are working/supported (ios, Max, JavaScript)
* if not - any plans too or dates?

Thanks!

tommyZZM, 2015-09-16:
[image]

lpommers, 2015-09-16:
I haven't been able to even connect my BB8 with computer via bluetooth (OSX) anyone had any luck?

mcgrews3, 2015-09-16:
I can detect the BB8 via bluetooth on OS X.  I just can't pair with it.

To 'detect' your BB8 you will need node.js and do a global install of cylon-ble.  The global install adds some command line tools.

https://github.com/hybridgroup/cylon-ble#commands

notably you will use `cylon-ble-scan` to scan nearby devices.  Note the ID of the BB8 and then you can use the info command to see information about the BB8 --> `cylon-ble-info <GUID>`

My guess is that BB8 is somewhere between Sphero and Ollie in terms of functionality.  The ollie apparently can be put into a debug mode by sending it some commands (set power level to x, blah blah).  For ollie this apparently done via the mobile app.  Sphero is more explicit in that you must pair the Sphero.

It could also be that because this is a Star Wars licensed product they may not be able to 'open it up'.  My hope was to make a better patrol/pathfinding app and just play around with it.

lpommers, 2015-09-16:
Yeah, one of the developers commented on this reddit thread about connecting to the BB8. He says that pairing it will cause it not to connect properly. 

I agree about the functionality, and the Star Wars licensed product bit.

https://www.reddit.com/r/Sphero/comments/3johbo/broken_bb8_failed_to_connect_to_3_separate_bt/cv1vmwj

mcgrews3, 2015-09-16:
Another note: I have use the cylon-ble node.js module to retrieve information from the BB8.  The problem is those services/characteristics don't have descriptions filled in - they just have a value of null.  I was not willing to test these as I imagine I could brick my BB8 by sending the wrong command.

I created a repo as I explore BB8.  Thanks.

https://github.com/mcgrews3/bb8-driver

Bjodol, 2015-09-17:
Hi! We want to program scripts for BB-8 within max one week. Is this doable?

mcgrews3, 2015-09-18:
In addition to creating this issue I also registered and created a support ticket with Sphero.  Recieved a response yesterday:

Sphero Customer Service, Sep 17, 2:40 PM:

Hello,

Thank you for reaching out to Sphero. At this time development is not supported and there is no estimate for an SDK.

-Cody

Mcgrews3, Sep 9, 2:05 PM:

Just got a new BB8 and am a node.js developer. Noticed there are resources and info about programming Ollie + Sphero but nothing on BB8.

Is development supported for BB8?

If so - what SDKs are working/supported (ios, Mac, JavaScript)?

if not - any plans too or dates?

Thanks!

JBX028, 2015-09-22:
I also raised a support ticket to sphero.com the first day I bought my BB8. Of course, I recieved the same response than you.
tickle (https://tickleapp.com/en-us/devices/) supports BB8 already but this is limited to just one platform and also limited comparing to what it would be possible to do with node.js along with a kinect for example.
This is a strategic error from my point of view. BB8 is more than just a simple sphero ball because he is cute that's why we want to add new functionalities on top of what is delivered out of the box.
Would be the opportunity for them to let the developers being creative and create the buzz even if this is already a success based on the number of BB8 bought already around the world.
I have activated some alerts based on key words and as soon as something seems possible, I will try immediatly.

JBX028, 2015-09-23:
Hi,

As expected, I have played with tickle on my iphone and I confirm that BB8 is natively supported. For example, there is component to move his head which is of course not a component for the standard sphero ball.
That being said, that mean the star wars license is probably not the reason why an sdk is not proposed yet.
To come back to tickle, this is definitely a nice tool to learn programming but impossible to create a sophisticated program. An option exists to export the scripts but the output is not readable with a text editor. No way to create a script with node.js in order to produce an input for tickle :-(...even if it would have been the case, we would be limited by the actions recognised by the app.

mcgrews3, 2015-09-29:
I can make the BB8 move and change colors but can't "read" data off of the BB8.

Looking at sphero's work - I assume the approach behaves like bluetooth classic - where a pipe is opened over bluetooth and data is streamed back and forth over the pipe.  Given BB8 and ollie use Bluetooth Low Energy (version 4??) this approach of point to point streaming is no longer valid.

The question is - how do they do reads and streaming on the newer devices (BLE) where GATT (Services and Characteristics) is used.

https://developer.bluetooth.org/TechnologyOverview/Pages/GATT.aspx

ibrahima, 2015-10-17:
@mcgrews3 do you "pair" with the BB-8 from your PC? Seems like BTLE works a little differently than standard bluetooth so I'm not sure what the procedure is. So the code in your repo actually allows you to drive the BB-8 via JS? That's awesome! Will check it out when I have time.

ali1234, 2015-10-17:
I have python code that can fully drive the BB-8 and read sensors and it probably works with Ollie too since they are virtually identical.

https://gist.github.com/ali1234/5e5758d9c591090291d6

mcgrews3, 2015-10-17:
First, this my opinion - I am a noob with bluetooth programming...

It doesn't pair per se.  It's more of a client - server relationship (or request + response if you like).  Yes, you can change the color and roll the robot around with the cylon adapter/device.  At this point it's pretty much an RC car.  With the sensor data it could graduate to robot.  I am thinking Sphero may downgrade to an earlier bluetooth version to stream the data.  Otherwise it is with BLE notifications.  Recall that ollie and sphero used the earlier bluetooth version where you basically have a hose between the phone and the robot - with both shooting data through.  BLE is more like "tickling" the robot "here" to get it to do X.

ali1234, 2015-10-17:
@mcgrews3 with Ollie and BB-8 the same old async serial protocol is forced through two GATT attributes. You write commands to one of them, the results come back from the other in the form of notifications. BLE notifications are asynchronous so you can ask for a continuous stream of data just like you could on the old system.

mcgrews3, 2015-10-17:
@ali123 Checking out your code now.  Wrote my previous post before seeing your post about efforts with python...

Thanks!!!

omeryagmurlu, 2015-12-05:
BB-8 and Ollie have the same Services and Characteristics. It shouldn't be any different in terms of bluetooth connections from Ollie. I wrote a javascript SDK for Ollie [here](https://github.com/omeryagmurlu/olliejs), it should work with BB-8 too.

halr9000, 2015-12-12:
👍

zikolach, 2015-12-14:
👍

jakswa, 2015-12-17:
For anyone stumbling through their first Bluetooth Low-Energy experience, like I was, I was able to get connected to my BB8 and move forward + change colors using these general steps: https://gist.github.com/jakswa/6d607ceb130ace7f3d0c

chr0n1x, 2015-12-18:
@jakswa pretty cool, will try soon. Thanks! :D

deadprogram, 2015-12-22:
It's almost here! We're working away at our hidden rebel base to give the people what you want.

```
$ PORT="F3:F2:6D:55:71:08" node examples/bluetooth-info.js                                                             
peripheral with ID f3f26d557108 found
bluetoothInfo:
  name: BB-7108000000000
  btAddress: f3f26d557108
  separator: 0
  colors: 0x0

ron@ron-XPS-13-9343:~/Development/sphero.js (feature/ble)
$ PORT="F3:F2:6D:55:71:08" node examples/color.js                                                             
peripheral with ID f3f26d557108 found
color 1
color 2
color 3
```

More very soon...

JBX028, 2015-12-22:
Hi,

Please check the cylon-bb8 driver created by @mcgrews3. Everything works fine and I have even exposed the actions (move, color, etc...) as web services with node-restify together with node-ngrok in order to connect my bb8 to ifttt. 
I really hope that one of the next framework update that is installed sometimes into the droid by the ios app, is not going to block everything as part of a new update...

doublecong123, 2015-12-23:
@jakswa pretty cool stuff. I am able to connect to bb8 through mac. The only thing is that after this exercise, I couldn't connect bb8 through iphone6 app. Do you see the same problem by any chance?
Thanks.

jakswa, 2015-12-23:
I did not experience that @doublecong123. I connected to it from android after shutting down my JS code (I remember it going red, like it does when it goes to sleep). Things have seemed fine, I was rolling it around earlier today as well.

deadprogram, 2015-12-24:
Hi, everyone.

The new master branch now includes full Ollie & BB-8 support. Please check it out, we need your feedback, and thanks for all the awesome community work here.

tlawrie, 2015-12-26:
Has anyone thought of connecting via node and syncing it up to IFTTT so that for example, when you get a message the BB-8 / Ollie will spin and light up?

rhencke, 2015-12-26:
@deadprogram Just tried this out - it works great.  There were several places my experience had to deviate from the instructions in README.md, though, that you may wish to be aware of:

* Using ```node ./node_modules/noble/examples/advertisement-discovery.js``` to determine the MAC address was not of much use, but it did help find the device ID which ended up working.  What my output ended up as was:

```
peripheral discovered (944f561f8cf441f3b5405ed48f5c63cf with address <unknown, unknown>, connectable true, RSSI -73:
	hello my local name is:
		BB-131D
	can I interest you in any of the following advertised services:
		[]
	here is my manufacturer data:
		"3330"
	my TX power level is:
		-18
```

* I could not pair to the BB-8 using Bluetooth, but it ended up being unnecessary.
* To connect to the BB-8, I had to supply my device's UUID instead of its MAC address:

```
var sphero = require("sphero"),
    bb8 = sphero("944f561f8cf441f3b5405ed48f5c63cf");
    // not bb8 = sphero("C8:48:52:3A:13:1D") which was the MAC address I eventually
    // figured out - that didn't work unfortunately.  But the UUID did.
```

tlawrie, 2015-12-26:
That's really great. Essentially followed @deadprogram guide to implementing that way yesterday. I know have something which runs from my laptop to the BB-8 droid.

My next step is associated that with an IFTTT recipe or something similar to turn the droid into an alarm / Notification Center. Any ideas on how that part would work?

My confusion at the moment is that if I expose a service up on a node cloud to IFTTT then it won't be able to reach my droid.

rhencke, 2015-12-26:
@tlawrie I'm toying with something similar, actually.  Let me know if you want to join forces - you can reach me at robert.hencke@gmail.com.  But, what you basically need is to have a computer in range with Bluetooth that can handle the notifications, and send them on to BB-8 for you.

kborgolte, 2015-12-26:
@rhencke connection with the UUID works fine! thx

olcar, 2015-12-27:
I can also confirm that most of the examples work fine on my brand new BB8 by using the UUID instead of MAC for Bluetooth connection.

I think that issue can now be close :)

deadprogram, 2015-12-27:
This is really cool, thanks everyone for the comments & feedback. I totally agree about the update to the OSX instructions being needed, I just pushed https://github.com/orbotix/sphero.js/commit/8430db7c18271bca4c3eaeabbf3ef6e0ddba8d7e please take a look everyone and see if this clarifies the connection instructions.

deadprogram, 2015-12-28:
Now that the module has been released, I am going to close this issue. Thank you so very much to everyone, and please open new issues with any specific requests or needs.

closed 2015-12-28 by deadprogram

olcar, 2015-12-28:
@deadprogram: many thanks to you!

dimorphic, 2015-12-29:
@deadprogram :heart_decoration:

mintuz, 2016-01-23:
@tlawrie check out something I've been working on which allows you to connect your BB8 to IFTTT. it runs a local web server on your machine, you just need to give access to it via your firewall and then use use ISP public IP Address to send requests to it.

https://github.com/mintuz/BB8-Commander/pull/7

I've not merged it yet into master but planning on doing that either today or tomorrow.

marcusletric, 2016-03-07:
Hi there! I want to say thanks for all the effort you've done creating this module, I like it very much, finally I can develop for my doird!

Just bought an ASUS USB-BT400 dongle and managed to connect my BB8 under WIN7
I needed some tweaks to do that so here are my findings how to do it under WIN7:

1. Install official driver that comes with the dongle
2. Download Zadig USB driver installer from here http://zadig.akeo.ie/
3. run the program, select options -> list all devices
4. select BCM20702A0 (the bluetooth device) from the dropdown
5. at driver section select WinUSB
6. click replace driver.

If all the node modules are installed, then you're ready to go.
I hope it is useful.