sphero-inc/sphero.js #27: Does this package support ollie?

MindlessRanger, 2015-09-05:
Does this package support ollie? Otherwise, are you working on making ollie a similar package?

omeryagmurlu, 2015-11-28:
I've written a similar (it is not capable of doing everything this SDK) SDK for Ollie [here](https://github.com/omeryagmurlu/olliejs/). Someone may want to take a look at it. Right now I'm writing a controller for Ollie with it.

deadprogram, 2015-12-24:
Hi, everyone.

The new `master` branch now includes full Ollie & BB-8 support. Please check it out!

mcgrews3, 2015-12-24:
Awesome!  Will check it out and thanks for the update!!!

deadprogram, 2015-12-27:
The conversation on #26 also includes a lot of Ollie, so I'd like to close this issue in favor of that one, if there are no objections.

closed 2015-12-28 by deadprogram