sphero-inc/sphero.js #28: Buffer parser doesn't work

simonas88, 2015-09-24:
Upon calling getBluetoothInof() or getPowerStateInfo() or any other methods that contain data in buffer format, the methods return buffer data instead of human readable data. 

For instance:
{ sop1: 255,
  sop2: 255,
  mrsp: 0,
  seq: 0,
  dlen: 33,
  data: <Buffer 53 70 68 65 72 6f 2d 47 52 50 00 00 00 00 00 00 36 38 38 36 65 37 30 34 64 63 37 65 00 47 52 50>,
  checksum: 47 }

deadprogram, 2015-12-02:
Hi, @simonas88 we found the problem with those commands, and it has been corrected in the `master` branch. We should be releasing an updated version to `npm` later today.

deadprogram, 2015-12-02:
Hello @simonas88 the new release is out, please give it try.

deadprogram, 2015-12-17:
Hi, @simonas88 did you have a chance to retry this?

simonas88, 2015-12-17:
Hello,

Unfortunately, I no longer work with the project that used the spheros nor
am I in possession of the devices. Which means that I can't test the fixes
in foreseeable future.

deadprogram, 2015-12-24:
OK, thanks for the feedback. Closing since it appears to be working now, thanks to all the recent commits.

closed 2015-12-24 by deadprogram