sphero-inc/sphero.js #31: node-gyp fails to compile during npm install

chr0n1x, 2015-11-01:
```
> serialport@1.7.1 install /home/kran/Code/orbotix/sphero.js/node_modules/serialport
> node-pre-gyp install --fallback-to-build

make: Entering directory `/home/kran/Code/orbotix/sphero.js/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
In file included from ../src/serialport.h:5:0,
                 from ../src/serialport.cpp:3:
../node_modules/nan/nan.h:261:25: error: redefinition of ‘template<class T> v8::Local<T> _NanEnsureLocal(v8::Local<T>)’
 NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Local<T> val) {
                         ^
../node_modules/nan/nan.h:256:25: error: ‘template<class T> v8::Local<T> _NanEnsureLocal(v8::Handle<T>)’ previously declared here
 NAN_INLINE v8::Local<T> _NanEnsureLocal(v8::Handle<T> val) {
                         ^
../node_modules/nan/nan.h:661:13: error: ‘node::smalloc’ has not been declared
     , node::smalloc::FreeCallback callback
             ^
../node_modules/nan/nan.h:661:35: error: expected ‘,’ or ‘...’ before ‘callback’
     , node::smalloc::FreeCallback callback
                                   ^
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Object> NanNewBufferHandle(char*, size_t, int)’:
../node_modules/nan/nan.h:665:50: error: ‘callback’ was not declared in this scope
         v8::Isolate::GetCurrent(), data, length, callback, hint);
                                                  ^
../node_modules/nan/nan.h:665:60: error: ‘hint’ was not declared in this scope
         v8::Isolate::GetCurrent(), data, length, callback, hint);
                                                            ^
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Object> NanNewBufferHandle(const char*, uint32_t)’:
../node_modules/nan/nan.h:672:67: error: call of overloaded ‘New(v8::Isolate*, const char*&, uint32_t&)’ is ambiguous
     return node::Buffer::New(v8::Isolate::GetCurrent(), data, size);
                                                                   ^
../node_modules/nan/nan.h:672:67: note: candidates are:
In file included from ../node_modules/nan/nan.h:25:0,
                 from ../src/serialport.h:5,
                 from ../src/serialport.cpp:3:
/home/kran/.node-gyp/4.2.1/include/node/node_buffer.h:31:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, v8::Local<v8::String>, node::encoding) <near match>
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/home/kran/.node-gyp/4.2.1/include/node/node_buffer.h:31:40: note:   no known conversion for argument 3 from ‘uint32_t {aka unsigned int}’ to ‘node::encoding’
/home/kran/.node-gyp/4.2.1/include/node/node_buffer.h:43:40: note: v8::MaybeLocal<v8::Object> node::Buffer::New(v8::Isolate*, char*, size_t) <near match>
 NODE_EXTERN v8::MaybeLocal<v8::Object> New(v8::Isolate* isolate,
                                        ^
/home/kran/.node-gyp/4.2.1/include/node/node_buffer.h:43:40: note:   no known conversion for argument 2 from ‘const char*’ to ‘char*’
In file included from ../src/serialport.h:5:0,
                 from ../src/serialport.cpp:3:
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Object> NanNewBufferHandle(uint32_t)’:
../node_modules/nan/nan.h:676:61: error: could not convert ‘node::Buffer::New(v8::Isolate::GetCurrent(), ((size_t)size))’ from ‘v8::MaybeLocal<v8::Object>’ to ‘v8::Local<v8::Object>’
     return node::Buffer::New(v8::Isolate::GetCurrent(), size);
                                                             ^
../node_modules/nan/nan.h: In function ‘v8::Local<v8::Object> NanBufferUse(char*, uint32_t)’:
../node_modules/nan/nan.h:683:12: error: ‘Use’ is not a member of ‘node::Buffer’
     return node::Buffer::Use(v8::Isolate::GetCurrent(), data, size);
            ^
make: *** [Release/obj.target/serialport/src/serialport.o] Error 1
make: Leaving directory `/home/kran/Code/orbotix/sphero.js/node_modules/serialport/build'
```

I *think* that there is probably something that I'm missing in my environment, but I can't find anything.

- g++ (Ubuntu 4.8.4-2ubuntu1~14.04) 4.8.4
- GNU Make 3.81

deadprogram, 2015-11-02:
Hi, @chr0n1x 

What version of Node.js are you running under? There are some known issues still with new versions of Node.js and node-pre-gyp.

chr0n1x, 2015-11-02:
Should be [this one](http://packages.ubuntu.com/trusty/nodejs)

What's the lowest stable version that works?

deadprogram, 2015-11-02:
I know node 0.10.28 works.

chr0n1x, 2015-11-02:
@deadprogram huh. lol ok I'll give that a shot, thank you!

artiintell, 2015-11-19:
Hello there,

I really need this to be FIXED! Any updates?

Have a big demo using cylon-sphero which would be a great ad also. Guys whoever is involved in moving this forward - your help on this is very important and much appreciated. 

Many Thanks

deadprogram, 2015-11-25:
Hi, everyone on this issue. The PR https://github.com/orbotix/sphero.js/pull/32 has been merged into the `master` branch, and a new module will be released to npm very soon. Thanks!

chr0n1x, 2015-11-25:
@deadprogram thank you for the update!

deadprogram, 2015-12-02:
Hi, @chr0n1x we released an updated version of this module last week, please give it a try, and reopen this issue if you have further troubles.

closed 2015-12-02 by deadprogram