sphero-inc/sphero.js #33: Broken dependencies

ea167, 2015-11-16:
package.json reference the exact version serialport@1.7.1 which DOES NOT COMPILE on Raspberry Pi, and breaks the whole Sphero package.

However if you npm install the latest version of serialport (2.0.5), it works well.

Referencing exact versions is BAD PRACTICE. 
Please correct your dependencies to allow updated versions!

deadprogram, 2015-11-25:
Hi, @ea167 the pull request https://github.com/orbotix/sphero.js/pull/32 has been merged into the `master` branch which updates the dependency.

A new module will be released to npm soon. Thanks!

deadprogram, 2015-12-02:
Hi, @ea167 we released an updated version of this module last week. Please give it a try, and reopen this issue if you have further troubles.

closed 2015-12-02 by deadprogram