sphero-inc/sphero.js #35: location.js sample always has undefined for position, velocity

bmidgley, 2015-12-02:
cd sphero.js/examples
node location.js
...
  xpos: undefined
  ypos: undefined
  xvel: undefined
  yvel: undefined
  sog: undefined

I printed out the object that does come back and it looks unrelated:

{ sop1: 255,
  sop2: 255,
  mrsp: 4,
  seq: 100,
  dlen: 1,
  data: <Buffer >,
  checksum: 150 }

I double-checked the command id and it really is going across as 0x15 now. This all lines up with the sphero docs I saw in http://docs-beta.gosphero.com/reference/

deadprogram, 2015-12-02:
The new release of this module is not out yet, @bmidgley should be later today.

deadprogram, 2015-12-02:
Hi, @bmidgley the release is out. Please give it try.

bmidgley, 2015-12-02:
Result is the same. I can confirm I have the latest. I added a roll command to the loop just in case that would help it register a change in location. No difference.

pi@rpi8gb ~ $ find -name packet.js -ls -or -name response.js -ls
316887   12 -rw-r--r--   1 pi       pi          11452 Nov 30 20:59 ./node_modules/sphero/lib/parsers/response.js
316877   12 -rw-r--r--   1 pi       pi           8816 Nov 30 20:59 ./node_modules/sphero/lib/packet.js
305427   12 -rw-r--r--   1 pi       pi          11452 Dec  1 16:45 ./sphero.js/lib/parsers/response.js
305425   12 -rw-r--r--   1 pi       pi           8816 Dec  1 16:45 ./sphero.js/lib/packet.js

deadprogram, 2015-12-02:
When I run the example called `location.js`, using the latest code, I get:

```
$ PORT=/dev/rfcomm1 node examples/location.js                                                                          
readLocator:
  xpos: 0
  ypos: 0
  xvel: 0
  yvel: 0
  sog: 0
readLocator:
  xpos: 0
  ypos: 0
  xvel: 0
  yvel: 0
  sog: 0
...
```

bmidgley, 2015-12-02:
I've tried on both an rpi (raspian) and ubuntu 14.04 laptop. Two different sphero 2s running 2.71 firmware. node is version 0.10.40 on the rpi. Same bad json object instead of locations.

what versions might be different in your environment?

└─┬ sphero@0.4.1
  ├── browser-serialport@2.0.2
  └─┬ serialport@2.0.5
    ├── async@0.9.0

deadprogram, 2015-12-02:
I'm also running Ubuntu 14.04 but using Node.js 5.0 (which should not matter). I'll test a couple different versions when I get back to workshop later today.

deadprogram, 2015-12-02:
Just tried with node 0.10.40, and same thing, worked fine for me:

```
$ node -v
v0.10.40
$ PORT=/dev/rfcomm1 node examples/location.js 
readLocator:
  xpos: 0
  ypos: 0
  xvel: 2
  yvel: 0
  sog: 2
readLocator:
  xpos: 0
  ypos: 0
  xvel: 0
  yvel: 0
  sog: 0
```

You might want to `rm -rf node_modules` and then `npm install` again?

deadprogram, 2015-12-02:
Also, you might need to update your Sphero to newer firmware. I have version:

```
msaVer: 3,
msaRev: 39
```

bmidgley, 2015-12-02:
It's also possible something broke in the newer firmware. This is what I have:

  msaVer: 3,
  msaRev: 71,

deadprogram, 2015-12-03:
How did you update your firmware to that version? Are you on iOS?

deadprogram, 2015-12-03:
The current iOS app is the only one that can update to that version of firmware. I updated mine to 3.71 and have now been able to reproduce your issue. 

Currently looking into this...

bmidgley, 2015-12-03:
Sorry yes you got it. I also used "sphero" app on ios. I'm glad we're seeing the same thing but I hope it wasn't intentionally removed from the new firmware.

deadprogram, 2015-12-03:
Hi @bmidgley 

I have it on good authority that the Location API will be working again as expected in firmware 3.73+ which will be out very soon.

In the meantime, if you have access to an Android device, you can "force update" the 3.59 firmware which seems to work fine.

Hope that helps!

bmidgley, 2015-12-04:
I was able to downgrade and confirm this works now!

Should these be parsed as some kind of "signed" rather than just unsigned "number"? eg I see 65535 rather than -1

bmidgley, 2015-12-10:
Would it be appropriate to add a signed type to the parser? This seems to cover this situation. I can make a PR.

diff --git a/lib/packet.js b/lib/packet.js
index a0c1217..4f4eee9 100644
--- a/lib/packet.js
+++ b/lib/packet.js
@@ -261,6 +261,7 @@ Packet.prototype._checkDSBit = function(ds, field) {
 
 Packet.prototype._parseField = function(field, data, pData) {
   var pField;
+  var width;
 
   data = data.slice(field.from, field.to);
   pField = utils.bufferToInt(data);
@@ -286,6 +287,12 @@ Packet.prototype._parseField = function(field, data, pData) {
     case "bitmask":
       pField = this._parseBitmaskField(pField, field, pData);
       break;
+    case "signed":
+      width = 8 * (field.to - field.from);
+      if (pField >= Math.pow(2, width - 1)) {
+        pField = pField - Math.pow(2, width);
+      }
+      break;
     default:
       this.emit("error", new Error("Data could not be parsed!"));
       pField = "Data could not be parsed!";
diff --git a/lib/parsers/response.js b/lib/parsers/response.js
index ccd3525..e3a8c14 100644
--- a/lib/parsers/response.js
+++ b/lib/parsers/response.js
@@ -348,13 +348,13 @@ module.exports = {
     fields: [
       {
         name: "xpos",
-        type: "number",
+        type: "signed",
         from: 0,
         to: 2
       },
       {
         name: "ypos",
-        type: "number",
+        type: "signed",
         from: 2,
         to: 4
       },

bmidgley, 2015-12-10:
Would it be appropriate to add a signed type? I can make a PR.

```javascript
diff --git a/lib/packet.js b/lib/packet.js
index a0c1217..4f4eee9 100644
--- a/lib/packet.js
+++ b/lib/packet.js
@@ -261,6 +261,7 @@ Packet.prototype._checkDSBit = function(ds, field) {
 
 Packet.prototype._parseField = function(field, data, pData) {
   var pField;
+  var width;
 
   data = data.slice(field.from, field.to);
   pField = utils.bufferToInt(data);
@@ -286,6 +287,12 @@ Packet.prototype._parseField = function(field, data, pData) {
     case "bitmask":
       pField = this._parseBitmaskField(pField, field, pData);
       break;
+    case "signed":
+      width = 8 * (field.to - field.from);
+      if (pField >= Math.pow(2, width - 1)) {
+        pField = pField - Math.pow(2, width);
+      }
+      break;
     default:
       this.emit("error", new Error("Data could not be parsed!"));
       pField = "Data could not be parsed!";
diff --git a/lib/parsers/response.js b/lib/parsers/response.js
index ccd3525..e3a8c14 100644
--- a/lib/parsers/response.js
+++ b/lib/parsers/response.js
@@ -348,13 +348,13 @@ module.exports = {
     fields: [
       {
         name: "xpos",
-        type: "number",
+        type: "signed",
         from: 0,
         to: 2
       },
       {
         name: "ypos",
-        type: "number",
+        type: "signed",
         from: 2,
         to: 4
       },
```

deadprogram, 2015-12-10:
That is for sure needed, and your PR greatly appreciated.

bmidgley, 2015-12-16:
https://github.com/orbotix/sphero.js/pull/36

closed 2015-12-22 by bmidgley