sphero-inc/sphero.js #38: setTimeout not working

JBX028, 2015-12-27:
Hi,

I have a stupid question but really I don't understand what's going on. By running an example from the official documentation (http://sdk.sphero.com/community-apis/javascript-sdk/), I wanted to let my bb8 to blink according the following script:

```javascript
bb8.connect(function() {
  var i = 0;
  do {
    setTimeout(function() {bb8.color("red");} , 1000);
    setTimeout(function() {bb8.color("green");} , 1000);
    i++
  } while (i<6);
});
```

....but bb8 blinks too fast and does not wait a second before to change the color.

atleastimtrying, 2015-12-27:
Both timeouts are set to 1000, try setting one to 2000. Also use a multiple of I. Exuse short text I'm on my phone.

JBX028, 2015-12-27:
Hi,
Thanks for the quick answer. Definitively not an issue with the sdk at all my apologies.... works fine with one of the 2 set to 2000. Now the "blink" is trigger just once and not multiple times despite the do/while but this is probably what you wanted to mention with "use a multiple of I"....not sure what you wanted to say here but I am going to close the issue anyway.

closed 2015-12-27 by JBX028

atleastimtrying, 2015-12-27:
Great it worked this may (or may not) help https://github.com/atleastimtrying/bb8-movement/blob/master/index.js

JBX028, 2015-12-27:
thanks @atleastimtrying !!