sphero-inc/sphero.js #40: Impossible to connect when 'connect ' is embedded into a function

JBX028, 2015-12-29:
Hi,

I would like to expose most of the commands (color, roll, etc, ....) as web services in order to control my device across various channels (ifttt, cortana, node-red, etc...)
Therefore, I have installed "restify" but bb8.connect is not fired when the command is embedded into a function. Below a simplified version of the source code to keep the most important elements:

```javascript
var sphero = require("sphero");
var bb8 = sphero("xxxx");

var restify = require("restify");

var server = restify.createServer({
  name: "bb8-rest-api",
  version: "1.0.0"
});

function bb8_SetColor(req, res, next) {
  bb8.connect(function() {
    console.log("connected");
    bb8.color(req.params.color);
  }
  res.send(200);
  return next();
});

server.get("/color/:color", bb8_SetColor);

server.listen("8081", function() {});
```

I have of course added several console.log and up to bb8.connect, the code is well executed and then nothing else happen after while the same code outside the function works fine.

Any ideas why the command is not recognised?

Thanks for any help you will provide

omeryagmurlu, 2015-12-29:
You should create an instance of sphero, right now you're just calling the class ```sphero``` as a function. Try using ```new```

dimorphic, 2015-12-31:
Can you try passing your `bb8` (sphero) reference to `bb8_setColor` function?
Shouldn't matter as code is valid anyway, but just a thought... :)

dimorphic, 2016-01-01:
Nevermind what i just said. I have the same problem.
Trying to trigger a sphero.connect() in another function and fails silently. :<

omeryagmurlu, 2016-01-01:
I used it in a nw.js application and connect method doesn't fail. Maybe something is wrong with restify?

JBX028, 2016-01-01:
Hi,
I will try to investigate deeper in detail this issue. Perhaps restify is the root cause and what I will do first   is to call the class from another framework. I'll post here the result of my investigation. Thanks.

JBX028, 2016-01-01:
Seems to be indeed an incompatibility with restify because bb8 is well visible even embedded into a function such as the following example:

```javascript
var sphero = require("sphero");
var bb8 = sphero("xxxx");

function test() {
  bb8.connect(function() {
    console.log('connected');
  });
}

test();
```

I have tried to debug my app with node-inspector but even if I can see the "connect" method triggered, it's complicated to understand the rest of the execution and understand the root cause.

JBX028, 2016-01-02:
I have tried today to separate restify and bb8 and used child_process to send remote commands from the parent to the child (https://nodejs.org/api/child_process.html#child_process_child_send_message_sendhandle_callback) but again, no connection.
I have then created a cli client for bb8 where we can pass parameters to the program and using "exec", I calls this program from the server. This solution works fine even if this is definitively not a proper solution but at least, in addition to the server, I will have a cli for bb8 that can be called from scripts.
Of course, I would prefer to have a better integration but I am happy to have something that work.