sphero-inc/sphero.js #43: Add `noble` and `serialport` to project dependencies

zaccolley, 2016-01-02:
Related to https://github.com/orbotix/sphero.js/pull/42.

Could the project require both dependencies and use config options to determine which device is being used?

This would probably simplify the installation process and make the examples clearer?

e.g
```JavaScript
device = sphero('########', { device: 'ollie' });
```

```JavaScript
device = sphero('########', { device: 'bb8' });
```

```JavaScript
device = sphero('########'); // falls back to original setup
```

deadprogram, 2016-01-02:
Thanks for the comment @zaccolley we cannot do this, due to not all machines (especially older Windows) having the BLE stack needed to install Noble. We want to allow those users to continue to use Sphero.js in their use case. Hence the current more "dynamic" dependency.

zaccolley, 2016-01-03:
OK gotcha

closed 2016-01-03 by zaccolley