sphero-inc/sphero.js #45: Possible to control BB8 Head?

mobyjames, 2016-01-04:
I was able to connect and control bb8 using standard commands, but am I correct in assuming that controlling the head is not supported? I'd be happy to investigate how to add it if someone could point me in the right direction. Is there a way to interface the motors directly?

rhencke, 2016-01-05:
The head is directly attached to the motor that drives the weight through a magnet mounted to a fixed stock - it is not a separate component you'd drive on its own.

This may help clarify things:

http://www.element14.com/community/blogs/linker/2015/09/16/disassembling-bb8-part-1

mobyjames, 2016-01-05:
Oh, perfect. Thanks for the clarification.

closed 2016-01-05 by mobyjames

debu66er, 2016-05-22:
@mobyjames did you get any code to move the head working?
i am trying to force bb8 to say YES but i am not able. I tried different combinations of roll and stop with no luck. Don't know if there is other functions to move the magnets inside for example. Thanks!

mobyjames, 2016-05-23:
As mentioned above, the head is not directly controlled. To get a fast nod / yes, I use the following (node.js):

```
var src = "motor 2 80 2 80\n" +
          "delay 50\n" +
          "motor 1 80 1 80\n" +
          "delay 50\n" +
          "motor 2 80 2 80\n" +
          "delay 50\n" +
          "motor 1 80 1 80\n" +
          "delay 50\n" +
          "motor 0 0 0 0\n";

actions['fast_nod'] = macros.compile(src);
```

debu66er, 2016-05-24:
Hi @mobyjames, thank you very much for your help.
I will try in a couple of day I have the BB8 again with me.
Meanwhile, if you know the translation of macros to current SDK functions... just to try to understand what is it really doing as I haven't found doc about macros (motor? delay?)... does it map with anything similar to this?

```
	bb8.roll(200,0);
	setTimeout(function() { bb8.roll(100,0); }, 50);
	setTimeout(function() { bb8.roll(200,0); }, 100);
	setTimeout(function() { bb8.roll(100,0); }, 150);
	setTimeout(function() { bb8.stop(); }, 200);
```
Again, thanks!