sphero-inc/sphero.js #48: Collision Detection Working

jhlau, 2016-01-08:
Hi,

I was just testing with collision detection and I am not getting asynchornous messages when my BB-8 has a collision. Has anyone tested if it works? I basically added a line to tell it to roll in the examples/collision-detection.js:

var sphero = require("../");
var orb = sphero(process.env.PORT);

    orb.connect(function() {
      orb.detectCollisions();
      orb.color("green");

      orb.on("collision", function(data) {
        console.log("collision detected");
        console.log("  data:", data);

        orb.color("red");

        setTimeout(function() {
          orb.color("green");
        }, 1000);
      }); 

      orb.roll(200, 0);

    });

No colour change, no console log output.

olcar, 2016-01-08:
I tried to test the collision detection as well and had to tweak the configuration thresholds in order to get events...

```
bb8.configureCollisions({
    meth: 0x01,
    xt: 0x20,
    yt: 0x20,
    xs: 0x20,
    ys: 0x20,
    dead: 0x50
  });
  bb8.on("collision", function(err, data) {
    if (err) {
      console.log(err);
    } else {
      console.log("collision detected!");
    }
  });
```

...those thresholds might be a little bit too sensitive but you should be able to test it.

deadprogram, 2016-01-08:
This makes me wonder if we should add additional key to indicate "preset" defaults per device. For example:

```
robot.configureCollisions({device: "bb8"});
```

which would then use default values such as those used by @olcar vs. the current defaults which are optimized for Sphero 2.0 devices.

What do you think?

olcar, 2016-01-08:
That would make sense. I wasn't really able to find the sweet spot for BB8, it probably needs further investigation but those settings might be a good starting point. Defining presets would definitely help us improve that as a community.

If I understand correctly, that means BB8 is not a Sphero 2.0 device and handles behaviours in a different manner?

If it's the case, what do you think about defining global device presets for other defaults settings like RotationRate, InactivityTimeout, etc. ?

deadprogram, 2016-01-08:
Most of these settings are the same, but the collision detection is one I've noticed a clear difference.

Anyhow, we'd love a PR that added this, or else I can work on it.

olcar, 2016-01-08:
OK, let's start with collision detection presets for BB8 then ;)

I can try to work it out during the weekend.

deadprogram, 2016-01-08:
Awesome! :bow:

jhlau, 2016-01-09:
I spent some time fiddling with the thresholds yesterday. I find that anything larger than 01h for 'dead' makes it very insensitive. I am currently using these thresholds:
    xt: 0x20,
    yt: 0x20,
    xs: 0x10,
    ys: 0x10,
    dead: 0x01

It's a little sensitive though. I'd get multiple collision events for a collision and I'm also geting collision events for rotating BB8 (in-place).

jhlau, 2016-01-11:
Any updates on this?

olcar, 2016-01-11:
I submitted a PR yesterday that was merged earlier today to add presets for BB8.

Can you test again on your side using the new example I added (collision-detection-bb8.js)?

Let me know if it works OK for you.

deadprogram, 2016-01-12:
The new 0.7.0 release includes this new code.

jhlau, 2016-01-12:
Yeap. Tested it a bit and it's working. It occasionally fails to detect some collision, but I think it's good enough.

Quick question - do the other SDKs have these sort of threshold values for BB8? I'd imagine it'd make sense to copy their threshold values.

deadprogram, 2016-01-20:
Hi, @jhlau I looked thru the other SDKs and was not able to find any specific default values per type of device. I suppose that means it is up to us to do so! :smile:

deadprogram, 2016-02-05:
@jhlau if you need further assist on this issue, please re-open it. Thanks again to you and to @olcar for the code!

closed 2016-02-05 by deadprogram