sphero-inc/sphero.js #49: ollie doesn't connect with uuid

codeNgamer, 2016-01-09:
I've got the following code which works for the most part, except that the callback is never called from
ollie.connect()

```javascript

const bleAddress = "e7:40:98:2a:20:8d";
noble = Npm.require('noble');
sphero = Npm.require('sphero');
// Cylon = Npm.require('cylon');
// Npm.require('cylon-ble');

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
    console.log("Scanning for ollie's in the area...");
    noble.startScanning();
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', function(peripheral) {
  if(peripheral.address === bleAddress) {
    console.log('Found your ollie!');

    if(peripheral.connectable) {
      console.log('I can connect to it');
      console.log('Attempting to connect now');

      noble.stopScanning();
      console.dir('device uuid is: ' + peripheral.uuid);

      ollie = sphero(peripheral.uuid);

      ollie.connect(function() {
        console.log('connected!');
      });
      ollie.on('error', function() {
        console.log('error');
      })
    }else {
      console.log('Womp womp, I cant connect to it right now');
    }

  }
});

```

deadprogram, 2016-01-09:
Hi, @codeNgamer not sure why you are calling the Noble code directly, you do not need to do any of that when using sphero.js, it does it for you.

In fact, another thing that sphero.js does is to convert a BLE address for the format "e7:40:98:2a:20:8d" to that needed by noble "e740982a208d" to connect to an Ollie.

The code you should be using is something like:

```
var sphero = require("sphero"),
    ollie = sphero("e7:40:98:2a:20:8d"); // change BLE address accordingly

ollie.connect(function() {
  console.log('connected!');
});
```

That's it, unless I am missing something?

codeNgamer, 2016-01-09:
That's what I was doing at first, but it wasn't working. I forgot to mention that the above code block gets executed server-side on the meteor framework so i wasn't sure if there was something different in it's ecosystem causing it to not work. So I figured i'd search, find it, then try to connect but it still doesn't work. Strange thing is, noble connects just fine.

deadprogram, 2016-01-09:
Since sphero.js is also scanning using Noble, having your own code that scans cannot work well at the present time. Once solution I have been working on, is being able to pass an existing Noble device connection into sphero.js.

codeNgamer, 2016-01-09:
I figured that also, but using this code below doesn't work either. I'll just continue to investigate

```
var sphero = require("sphero"),
    ollie = sphero("e7:40:98:2a:20:8d"); // change BLE address accordingly

ollie.connect(function() {
  console.log('connected!');
});
```

deadprogram, 2016-01-09:
Does not work inside your Meteor app? Can you share a little more context of how you call this? Also what OS/Node version, please?

codeNgamer, 2016-01-09:
So i'm running it on osx/el capitan, meteor 1.2.1 which is based on node 0.10.40

deadprogram, 2016-01-09:
How are you calling the code from your Meteor app? Sorry if my questions seem silly, but I am not very familiar with Meteor, having only played with it a little when it first came out.

codeNgamer, 2016-01-09:
No that's fine. Basically, I pull request all npm packages needed by the meteor project with the code below:

```javascript
 Npm.depends({
  noble: '1.2.1',
  sphero: '0.6.0',
  spheron: '0.0.3',
  cylon: '1.2.0',
  'cylon-ble': '0.9.0',
  'cylon-ollie': '0.8.0',
});
```

then use the below code:

```javascript
var sphero = require("sphero"),
    ollie = sphero("e7:40:98:2a:20:8d"); // change BLE address accordingly

ollie.connect(function() {
  console.log('connected!');
});
```
which is executed from a script loaded server side

deadprogram, 2016-01-15:
Hi, @codeNgamer 

When using the latest release of sphero.js, you should be able to the the Noble peripheral from your code:

```
ollie = sphero(peripheral.uuid, {peripheral: peripheral});
```

Please give it a try!

deadprogram, 2016-01-20:
Hi, @codeNgamer did you have a chance to give a it try yet? Please let us know.

codeNgamer, 2016-01-25:
hey @deadprogram sorry for the late reply. I did try that, briefly though. it didn't work but i havent tested yet, been super busy at work. I'll drop feedback soon

b44p, 2016-02-05:
Having same issues unable to connect using uuid

deadprogram, 2016-02-05:
@b44p on OS X you use the UUID, on Linux you use the BLE address. Make sure you are entering the correct value as returned by the BLE scan using Noble.

b44p, 2016-02-05:
Thanks @deadprogram. I am on OSX and am using cylon-ble-scan to get the uuid. I get a 32 character string.
I then use the following code to connect and do something but nothing happens.
What am I missing here ? 

```
var Cylon = require('cylon');

Cylon.robot({
  connections: {
    bluetooth: { adaptor: 'central', uuid: 'UUID', module: 'cylon-ble' }
  },

  devices: {
    ollie: { driver: 'ollie', module: 'cylon-sphero-ble' }
  },

  work: function(my) {
    my.ollie.color(0x00FFFF);

    after(500, function() {
      my.ollie.color(0xFF0000);
    });

    after(1000, function() {
      my.ollie.roll(60, 0);
    });

    after(2000, function() {
      my.ollie.roll(60, 180);
    });

    after(3000, function() {
      my.ollie.stop();
    });
  }
}).start();
```

deadprogram, 2016-02-05:
This may seem like a silly question, but you are using `uuid: 'Your actual UUID here'` in your code, right?

b44p, 2016-02-05:
Haha - you know the pain of those brain farts too well it seems - but no - i made sure i plugged in my 32 character UUID that i refrain to disclose for an unfounded fear that my ollie could be hacked remotely

deadprogram, 2016-02-06:
Hi, @b44p seems like this would be more correctly an issue entered on https://github.com/hybridgroup/cylon-sphero-ble would you mind opening one on that repo?

b44p, 2016-02-12:
Hey @deadprogram - I realize that I am not putting the ollie into dev mode but according to [@peterrainbow](http://community.gosphero.com/discussion/109/hacking-the-ollie-from-raspberry-pi-and-node) where he created cylon-ollie. He appears to say that there he updated https://github.com/peterainbow/cylon-ollie to programmatically turn dev mode on. Am i missing something here ?

deadprogram, 2016-02-12:
That is all auto handled by the current version of Sphero.js so no need.

b44p, 2016-02-12:
I was able to use advertisement_discovery and get Ollie's UUID and then do this. Does not work.
```
var sphero = require("sphero"),
    bb8 = sphero("2bb63049a903424fb602fff64cfb3fdc"); // This my OLLIE

bb8.connect(function() {
  // roll BB-8 in a random direction, changing direction every second
  setInterval(function() {
    var direction = Math.floor(Math.random() * 360);
    bb8.roll(150, direction);
  }, 1000);
});

```

deadprogram, 2016-02-13:
Hi, @b44p shouldn't you be using the 32-char UUID instead of the BLE address in the code above, since you are on OSX? I booted up my OS X machine (still running Yosemite) installed the latest Sphero.js, and was able to run without troubles.

b44p, 2016-02-13:
Hey @deadprogram - tried both - no joy :-1:

deadprogram, 2016-02-13:
I will install El Capitan and dig in further...

b44p, 2016-02-15:
Thank you kind sir

deadprogram, 2016-02-25:
Hi, @b44p still researching on this. Another question: what exact Mac do you have? Including the specific Bluetooth adaptor, please.

b44p, 2016-03-03:
argh sorry for the late response - Im using 2 macs

**MAC1**

 

>  Model Name:	iMac
>   Processor Name:	Intel Core i5
>   Processor Speed:	3.4 GHz
>   Number of Processors:	1
>   Total Number of Cores:	4
>   L2 Cache (per Core):	256 KB
>   L3 Cache:	6 MB
>   Memory:	32 GB
> 
**> Apple Bluetooth Software Version:	4.4.3f4 16616**
>   Hardware, Features, and Settings:
>   Name:	1PP-16
>   Address:	88-63-DF-A0-4D-0C
>   Bluetooth Low Energy Supported:	Yes
>   Handoff Supported:	Yes
>   Instant Hot Spot Supported:	Yes
>   Manufacturer:	Broadcom
>   Transport:	USB
>   Chipset:	20702B0
>   Firmware Version:	v112 c9036
>   Bluetooth Power:	On
>   Discoverable:	Off
>   Connectable:	Yes
>   Auto Seek Pointing:	On
>   Remote wake:	On
>   Vendor ID:	0x05AC
>   Product ID:	0x828D
>   HCI Version:	0x6
>   HCI Revision:	0x234C
>   LMP Version:	0x6
>   LMP Subversion:	0x4170
>   Device Type (Major):	Computer
>   Device Type (Complete):	Mac Desktop
>   Composite Class Of Device:	0x380104
>   Device Class (Major):	0x01
>   Device Class (Minor):	0x01
>   Service Class:	0x1C0
>   Auto Seek Keyboard:	On
> 

**MAC 2**

> Model Name:    MacBook Air
>  Model Identifier:    MacBookAir6,2
>  Processor Name:    Intel Core i5
>  Processor Speed:    1.3 GHz
>  Number of Processors:    1
>  Total Number of Cores:    2
>  L2 Cache (per Core):    256 KB
>  L3 Cache:    3 MB
>  Memory:    8 GB
>  Boot ROM Version:    MBA61.0099.B21
>  SMC Version (system):    2.13f15
> 
**> Apple Bluetooth Software Version:    4.4.3f4 16616**
>  Hardware, Features, and Settings:
>  Name:    Varun’s MacBook Air
>  Address:    84-38-35-64-E4-B3
>  Bluetooth Low Energy Supported:    Yes
>  Handoff Supported:    Yes
>  Instant Hot Spot Supported:    Yes
>  Manufacturer:    Broadcom
>  Transport:    USB
>  Chipset:    20702B0
>  Firmware Version:    v112 c9037
>  Bluetooth Power:    On
>  Discoverable:    On
>  Connectable:    Yes
>  Auto Seek Pointing:    On
>  Remote wake:    On
>  Vendor ID:    0x05AC
>  Product ID:    0x828F
>  HCI Version:    0x6
>  HCI Revision:    0x234D
>  LMP Version:    0x6
>  LMP Subversion:    0x4170
>  Device Type (Major):    Computer
>  Device Type (Complete):    Mac Portable
>  Composite Class Of Device:    0x38010C
>  Device Class (Major):    0x01
>  Device Class (Minor):    0x03
>  Service Class:    0x1C0
>  Auto Seek Keyboard:    On

deadprogram, 2016-03-04:
The `bluetooth-hci-socket` warning is just telling you "this optional module is not being installed". If you were installing on Linux or Windows, that module would be installed, but you would see a warning for `xpc-connection` not being installed, since `xpc-connection` is Mac-only.

@b44p we have an issue open on Noble itself here: https://github.com/sandeepmistry/noble/issues/340 you might want to chime in.