sphero-inc/sphero.js #50: Can't pair BB-8 with OSX 10.11.2

Myeris, 2016-01-09:
Hey,

I'm trying to pair my BB-8 with my Macbook and I can't seem to find how to.
I've followed every step, installed noble, installed sphero SDK, got my BB-8 UUID, no problem there.

Then I tried to pair it and this is where I'm stuck. First of, it seems my Macbook doesn't not support BLE connection since it was never able to detect the BB-8. I downloaded LightBlue on the AppStore and I was able to see the BB-8 and connect it.

However, I can seem to find the tty.Sphero thingy.
When I type this command `ls -a /dev | grep tty.Sphero`, it returns nothing.

Any ideas?

deadprogram, 2016-01-09:
Hi, @Myeris you do not pair with BLE devices the same way you do with Bluetooth Classic devices. Please try following the instructions here: https://github.com/orbotix/sphero.js#os-x

Myeris, 2016-01-09:
Oh right... Sorry I didn't read as thoroughly as I thought. Thanks for your time and answer.

closed 2016-01-09 by Myeris