sphero-inc/sphero.js #51: readLocator command response data

okoriko, 2016-01-10:
Hi!

I've got some troubles with the readLocator response data.

Let's say I set the configuration as:

```
  var opts = {
    flags: 0x00,
    x: 0x00FF, // x = 255 cm
    y: 0x0000,
    yawTare: 0x0
  };

  orb.configureLocator(opts);
```

I will 2 as `xpos` value the when I do:
```
    orb.readLocator(function(err, data) {
      if (err) {
        console.log("error: ", err);
      } else {
        console.log("readLocator:");
        console.log("  xpos:", data.xpos);  // xpos is 2
        console.log("  ypos:", data.ypos);
        console.log("  xvel:", data.xvel);
        console.log("  yvel:", data.yvel);
        console.log("  sog:", data.sog);
      }
    });
```

In parallel though, if I read the async packets, I can get the real value:
```
  orb.on("async", function(response){
    var xField = {from: 0, to: 2, type: 'signed'};
    var yField = {from: 2, to: 4, type: 'signed'};
    var x = Packet.prototype._parseField(xField, response.data); // x is 255
    var y = Packet.prototype._parseField(yField, response.data);
    console.log('pos:', x, y);
  });
```

I've looked at the response from ollie on the readLocator and the response itself looks wrong. It has the xpos set to 2 and x vel is to 25 (`0x19`):
```
<Buffer ff ff 00 01 0b 00 02 00 00 00 19 00 00 00 00 d8>
```

Not sure whether it matters but I am using a Ollie.

Does anyone know what might be going on, why am I getting the wrong data in the readLocator but correct position in the streaming/async packets?

okoriko, 2016-01-10:
Quick note: it looks like the iOS SDK is implementing the locator support using "odometer" and "velocity" data streaming:
https://github.com/orbotix/Sphero-iOS-SDK/blob/master/samples/Locator/Locator/ViewController.m#L100

And by reading the position from the async packets:
https://github.com/orbotix/Sphero-iOS-SDK/blob/master/samples/Locator/Locator/ViewController.m#L110

It might be the way to go with this? 
It still does not explain the behavior on the readlocator response though.

jhlau, 2016-01-10:
This might not answering your question, but I am calling the odometer to the x and y positions [1]. I was fiddling with readLocator before but found that they gave the same values for xpos and ypos, but the odometer way is just simpler.

[1] https://github.com/orbotix/sphero.js/blob/master/examples/stream-odometer.js

deadprogram, 2016-01-11:
There was a previous issue https://github.com/orbotix/sphero.js/issues/35 that might be related? What version of firmware are you running?