sphero-inc/sphero.js #56: Noble dependency says it's missing when it's not

brianfeister, 2016-01-14:
So I realize that I need the additional npm dependency `noble`, which I've installed locally (do I need it globally installed)?  I am trying to run a simple script from the examples, as follows:
```
// test.js
var sphero = require("sphero"),
    bb8 = sphero("D6:35:5E:95:BA:A2"); // change BLE address accordingly

bb8.connect(function() {
  // roll BB-8 in a random direction, changing direction every second
  setInterval(function() {
    var direction = Math.floor(Math.random() * 360);
    bb8.roll(150, direction);
  }, 1000);
});
```
 Despite the fact that I have `noble` installed and in `node_modules`, I'm getting the following:
```
➜  sphero git:(master) ✗ node test.js                                                 
It looks like you want to connect to a Sphero BB-8 or Sphero Ollie,
but did not install the 'noble' module.

To install it run this command:
npm install noble

For more information go to https://github.com/sandeepmistry/noble
/Users/brianfeister/dev/node_modules/sphero/lib/adaptors/ble.js:32
    throw new Error("Missing noble dependency");
          ^
Error: Missing noble dependency
    at initBLE (/Users/brianfeister/dev/node_modules/sphero/lib/adaptors/ble.js:32:11)
    at new Adaptor (/Users/brianfeister/dev/node_modules/sphero/lib/adaptors/ble.js:51:5)
    at Object.load (/Users/brianfeister/dev/node_modules/sphero/lib/loader.js:29:10)
    at new Sphero (/Users/brianfeister/dev/node_modules/sphero/lib/sphero.js:52:44)
    at sphero (/Users/brianfeister/dev/node_modules/sphero/index.js:21:10)
    at Object.<anonymous> (/Users/brianfeister/dev/sphero/test.js:2:11)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
➜  sphero git:(master) ✗ npm install noble
npm WARN optional dep failed, continuing bluetooth-hci-socket@0.4.2

> xpc-connection@0.1.4 install /Users/brianfeister/dev/sphero/node_modules/noble/node_modules/xpc-connection
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
  SOLINK_MODULE(target) Release/binding.node
noble@1.3.0 node_modules/noble
├── bplist-parser@0.0.6
├── debug@2.2.0 (ms@0.7.1)
└── xpc-connection@0.1.4 (nan@2.2.0)
➜  sphero git:(master) ✗ node test.js     
It looks like you want to connect to a Sphero BB-8 or Sphero Ollie,
but did not install the 'noble' module.

To install it run this command:
npm install noble

For more information go to https://github.com/sandeepmistry/noble
/Users/brianfeister/dev/node_modules/sphero/lib/adaptors/ble.js:32
    throw new Error("Missing noble dependency");
          ^
Error: Missing noble dependency
    at initBLE (/Users/brianfeister/dev/node_modules/sphero/lib/adaptors/ble.js:32:11)
    at new Adaptor (/Users/brianfeister/dev/node_modules/sphero/lib/adaptors/ble.js:51:5)
    at Object.load (/Users/brianfeister/dev/node_modules/sphero/lib/loader.js:29:10)
    at new Sphero (/Users/brianfeister/dev/node_modules/sphero/lib/sphero.js:52:44)
    at sphero (/Users/brianfeister/dev/node_modules/sphero/index.js:21:10)
    at Object.<anonymous> (/Users/brianfeister/dev/sphero/test.js:2:11)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
➜  sphero git:(master) ✗ 
```

deadprogram, 2016-01-14:
Hi @brianfeister you are on OSX from the looks of it. What version are you running? Also, what node version?

brianfeister, 2016-01-14:
WOW! Thanks for the quick reply @deadprogram.  I guess it could be noted in the docs that if you clone this repo, `sphereo` will fail to install when running `npm install sphero noble` because it would be a circular dependency.  So, cloning this repo and using it as a sandbox becomes a bit confusing.  I created an empty sandbox directory and was able to get running right away with just `npm install sphero noble`.

Sorry for the side question, but it seems unusual to me that I was only able to connect via the uuid that I got back from the `node ./node_modules/noble/examples/advertisement-discovery.js` command in the docs, but not the MAC address that I got from the details inside the iPhone app... but this one is probably just my own error / misunderstanding.

closed 2016-01-14 by brianfeister

deadprogram, 2016-01-14:
Hi @brianfeister when connecting on OSX, Noble requires that you use the "uuid' instead of the Bluetooth address. I'm not sure why that is the case within the OSX APIs, I suppose due to the "iBeacon" set of wrappers. In any case, we should probably add some separate script to the module, to make it easier to figure out the correct address.

ShahNewazKhan, 2017-02-21:
hi @deadprogram I was having this same issue, I created an Electron app and added the sphero npm module. 

Here is my envoriment on OSX 10.11:

```
$ node -v
v7.2.0

$ electron -v
v1.4.10
```
I was getting this error thrown from the initBle function:

```javascript
function initBLE() {
  var isChrome = typeof chrome !== "undefined";

  // thanks to https://github.com/jgautier/firmata/blob/master/lib/firmata.js
  try {
    if (isChrome) {
      // browser BLE interface here...
      console.error("Browser-based BLE interface is not yet supported.");
    } else {
      ble = require("noble");
    }
  } catch (error) {
    ble = null;
  }

  if (ble == null) {
    var err = [
      "It looks like you want to connect to a Sphero BB-8 or Sphero Ollie,",
      "but did not install the 'noble' module.", "",
      "To install it run this command:",
      "npm install noble", "",
      "For more information go to https://github.com/sandeepmistry/noble"
    ].join("\n");

    console.error(err);
    throw new Error("Missing noble dependency");
  }
}
```

This meant that the else clause ` ble = require("noble") ` was throwing an error, when I caught this error in console, it was throwing:

`Module version mismatch. Expected 50, got 51.`

I was able to resolve it by running the following from [the Electron documentation](http://electron.atom.io/docs/tutorial/using-native-node-modules/)

```
npm install --save-dev electron-rebuild

# Every time you run "npm install", run this:
./node_modules/.bin/electron-rebuild

# On Windows if you have trouble, try:
.\node_modules\.bin\electron-rebuild.cmd
```