sphero-inc/sphero.js #62: Access to heading value from BB8 ?

reidakdumont, 2016-02-05:
Hi,

Is it possible to have access to the heading value of bb8 ? I search for this functionality but I find nothing to get it and it could be useful to have it.

Thanks

deadprogram, 2016-02-05:
It is typical to use the Inertial Measurement Unit (IMU) for this info. Please check out the example code https://github.com/orbotix/sphero.js/blob/master/examples/stream-imu-angles.js

Hope that helps!

reidakdumont, 2016-02-08:
Hi deadprogram,

yes it helped me but now I have a little because the add of the call to streamImuAngles add some sort of lag in the command executing for bb8. Do you know if there is a solution to decrease the framerate of sending data to minimize the add of lag ?

Thanks

deadprogram, 2016-02-10:
Perhaps you want to play with different values for the `timeout` value for sending commands. For example:

```
var bb8 = new Sphero("F3:F2:6D:55:71:09", { timeout: 300 });
```

Hope that helps!

deadprogram, 2016-02-10:
From now on, I am going to use "HTH" for "Hope that helps!" to save on bytes. :P

deadprogram, 2016-02-25:
Hi, @reidakdumont did changing the timeout value help you? Please let us know.