sphero-inc/sphero.js #64: Cannot open /dev/rfcomm0 - Ubuntu

MartinT86, 2016-02-14:
Hi,

I'm an Ubuntu newbie and I'm struggling to use the sphero package.
Using your main example I am just getting the following error;
Cannot open /dev/rfcomm0

I have connected the sphero using bluetooth manager and it says the device is connected.
I have also added the following to a new file called - /etc/udev/rules.d/55-rfcomm.rules;
KERNEL=="rfcomm[0-9]*", NAME="%k", GROUP="dialout"

Any help would be great.

Thanks

deadprogram, 2016-02-14:
You might not be a member of the `dialout` group, run `whoami` and  `groups`. One way to test permission problems, is to run your code under sudo.

Hope that helps!

MartinT86, 2016-02-14:
thanks for the fast reply!

yes, looks like it was a permission issue. I'm now running it as "sudo node app.js" and it runs.

thanks again

commit 0a4b22e5 (2016-02-23): Fix #64 by adding recommendation to README to run under 'sudo' when encountering Linux permission problem
closed 2016-02-23 by deadprogram