sphero-inc/sphero.js #69: Connection Issue

DarrenGuinness, 2016-04-04:
I am working to get the Ollie up and running on a windows 10 machine. I have installed node, node-gyp, noble, sphero noble, and their dependencies. I can successfully find Bluetooth connections via ./advertisement-discovery.js, and can connect to peripherals using ./peripheral-explorer.js, however when I run the basic connection example with the Ollie's address, I get the following output:

C:\Users\IronLab\Desktop\test\node_modules\sphero\lib\adaptors\ble.js:404
  return this._connectedService(serviceId).characteristics;
                                          ^

TypeError: Cannot read property 'characteristics' of null
    at Adaptor._connectedCharacteristics (C:\Users\IronLab\Desktop\test\node_modules\sphero\lib\adaptors\ble.js:404:43)
    at Adaptor._connectCharacteristic (C:\Users\IronLab\Desktop\test\node_modules\sphero\lib\adaptors\ble.js:357:12)
    at C:\Users\IronLab\Desktop\test\node_modules\sphero\lib\adaptors\ble.js:302:12
    at Adaptor._connectService (C:\Users\IronLab\Desktop\test\node_modules\sphero\lib\adaptors\ble.js:338:5)
    at C:\Users\IronLab\Desktop\test\node_modules\sphero\lib\adaptors\ble.js:300:10
    at Adaptor._connectBLE (C:\Users\IronLab\Desktop\test\node_modules\sphero\lib\adaptors\ble.js:324:5)
    at Adaptor.getCharacteristic (C:\Users\IronLab\Desktop\test\node_modules\sphero\lib\adaptors\ble.js:299:8)
    at Adaptor.writeServiceCharacteristic (C:\Users\IronLab\Desktop\test\node_modules\sphero\lib\adaptors\ble.js:274:8)
    at Adaptor.setTXPower (C:\Users\IronLab\Desktop\test\node_modules\sphero\lib\adaptors\ble.js:203:8)
    at C:\Users\IronLab\Desktop\test\node_modules\sphero\lib\adaptors\ble.js:156:10

Am I missing an extra dependency?

deadprogram, 2016-04-05:
Hi @DarrenGuinness what is your output from `/advertisement-discovery.js` for your Ollie?

Also, are you able to run `peripheral-explorer.js` using the Ollie's ID? What is the output?

DarrenGuinness, 2016-04-05:
advertisement-discovery output: 

node ./advertisement-discovery.js
peripheral discovered (ed9a8071f1fd with address <ed:9a:80:71:f1:fd, random>, connectable true, RSSI -59:
        hello my local name is:
                2B-F1FD
        can I interest you in any of the following advertised services:
                ["22bb746f2ba075542d6f726568705327"]
        my TX power level is:
                -10

peripheral discovered (ffc7d9af5def with address <ff:c7:d9:af:5d:ef, random>, connectable true, RSSI -67:
        hello my local name is:
                2B-5DEF
        can I interest you in any of the following advertised services:
                ["22bb746f2ba075542d6f726568705327"]
        my TX power level is:
                -10

peripheral discovered (1c1ac09ee2c5 with address <1c:1a:c0:9e:e2:c5, public>, connectable true, RSSI -74:
        hello my local name is:
                undefined
        can I interest you in any of the following advertised services:
                []
        here is my manufacturer data:
                "4c000906031cc0a81b53"

--------------------------End of output----------------------------

I am able to run the peripheral-explorer script using the Ollie id. The output is as follows:

node ./peripheral-explorer.js ed9a8071f1fd
peripheral with ID ed9a8071f1fd found
  Local Name        = 2B-F1FD
  TX Power Level    = -10
  Service Data      =
  Service UUIDs     = 22bb746f2ba075542d6f726568705327

services and characteristics:
--------------------------End of output----------------------------

deadprogram, 2016-04-05:
Hi @DarrenGuinness I see that no services/characteristics were found for that ID, similar to the problem here https://github.com/orbotix/sphero.js/issues/49 except you are on Windows. Hmm.

Which Bluetooth LE adaptor are you using?

deadprogram, 2016-04-05:
Hello @DarrenGuinness I saw this issue on Noble https://github.com/sandeepmistry/noble/issues/372 perhaps you can try downgrading bluetooth-hci-socket to v0.4.2 as suggested there?

debu66er, 2016-05-21:
Hi @deadprogram I am suffering exactly the same issue on Raspberry Pi (with Raspbian Jessi). I downgraded bluetooth-hci-socket to v0.4.2 but it continues failing with same error.
@DarrenGuinness, did it work for you?

DarrenGuinness, 2016-05-24:
I ended up switching to ubuntu 14.04. After that the setup and install was a cinch.

voyager44, 2016-09-29:
@deadprogram , @debu66er 

Hi
I am having exactly the same problem/stacj  on Rpi3 -- here is the output when I run peripheral-explorer -- I installed sphero and noble as of today, _even though unlike @DarrenGuinness I do get a non-empty "services and characteristics" --_   -- any suggestions I may want to try?

sudo node peripheral-explorer.js "e3b3fe9c5254"
peripheral with ID e3b3fe9c5254 found
  Local Name        = BB-5254
  TX Power Level    = 6
  Manufacturer Data = 3330
  Service Data      = 
  Service UUIDs     = 22bb746f2ba075542d6f726568705327

services and characteristics:
1800 (Generic Access)
1801 (Generic Attribute)
  2a00 (Device Name)
    properties  read
    value        | ''
  2a01 (Appearance)
    properties  read
    value        | ''
  2a04 (Peripheral Preferred Connection Parameters)
    properties  read
    value        | ''