sphero-inc/sphero.js #74: Any way to reset BB8?

debu66er, 2016-05-18:
Hi, playing around with BB8 and this SDK and sometimes, it seems BB8 gets hanged. Usually it is a matter of time, or sometimes if I start the official mobile app it starts working again.

Is there any function in the SDK to reset, initialize or anything similar to try to fix BB8 from my code? I took a look but I didn't find any obvious function name.

Any help or comment? Anybody suffer same random hang?

Thanks in advance,

     Chemi.

debu66er, 2016-05-22:
At the end, I am not able to explain when this issue happens. It seems some how random.
But I was able to find a not scientific procedure to recover it which is connect() and disconnect() several times from my app.

What I see is that when connecting, BB8 executes the commands that failed before. Is like it has memory and as it didn't answered before it is trying to answer now. So after I connect and disconnect several times it seems I clean its queue or whatever it has and it starts working again.

Any function to clean, reset or whatever it needs to start from scratch?

debu66er, 2016-05-24:
Finally, a ping() call after connect() function avoided any more hanged.

closed 2016-05-24 by debu66er

jmvandamme, 2017-02-26:
Hi, could you explain what you mean with ping() call after connect() ? thx.

debu66er, 2017-03-01:
I can't explain why this fixed my problem.... but this is the code:

bb8.connect(function() 
{
	console.log("##### Connected to BB8");
	bb8.ping(); // Workaround for BB8 hangs?
});

ShahNewazKhan, 2017-03-21:
@jmvandamme were you able to resolve the issue with a ping() call?

jasherone, 2017-11-24:
Hi, it looks like I have very common issue. **Did you manage to fix that?**

I am trying to use this https://github.com/charliegerard/leap_sphero.git

I tested on different versions of MacOS, but I still have the same issue.

**Fixes tried**
1/ I found that the issue could be caused by noble version 1.8.0. I downgraded noble to 1.7.0, but I still have the issue.

2/ Call ping() after connect() https://github.com/orbotix/sphero.js/issues/37

**Current Config**
Mac OSX 10.12.6
Sphero model: S003, SPRK (not plus)
   "express": "^4.16.2",
    "leapjs": "^0.6.4",
    "noble": "^1.7.0",
    "serialport": "^3.1.2",
    "sphero": "^0.9.2"

**Issue**
I only send roll commands to sphero, and I limit them to one command every 200ms.

Fist few commands execute instantly, but then after 10-20 seconds (about 30-50 commands), they suddenly they start lagging up to few seconds long lags and I was getting following message

Unhandled rejection Error: Command sync response was lost.
    at Sphero.handler (/Users/martin/Workspace/LEAP Sphero MJ/node_modules/sphero/lib/sphero.js:252:21)
    at ontimeout (timers.js:475:11)
    at tryOnTimeout (timers.js:310:5)
    at Timer.listOnTimeout (timers.js:270:5)

I tried sending commands more often and less often, it made no difference.

And I also tried waiting (for example 60 seconds) and then started sending commands. It looks like Sphero starts to lag 10-20 seconds after I start to send commands no matter how many commands I send and how much I waited before.

_The error line is in this funcion:_

/**
 * Adds a promise to the queue, to be executed when a response
 * gets back from the sphero.
 *
 * @private
 * @param {Array} cmdPacket the bytes array to be send through the wire
 * @param {Function} resolve function to be triggered on success
 * @param {Function} reject function to be triggered on failure
 * @example
 * sphero._execCommand(packet, resolve, reject);
 * @return {void}
 */
Sphero.prototype._queuePromise = function(cmdPacket, resolve, reject) {
  var seq = cmdPacket[4];
  var handler = function(err, packet) {
    clearTimeout(this.responseQueue[seq].timeoutId);
    this.responseQueue[seq] = null;
    this.busy = false;
    if (typeof resolve === "function") {
      if (!err && !!packet) {
        resolve(packet);
      } else {
      var error = new Error("Command sync response was lost.");
        reject(error);
      }
    }
    this._execCommand();
  };