sphero-inc/sphero.js #77: Diconnect function turns on BB8 light

debu66er, 2016-05-22:
When I close my app, I disconnect before from BB8 with disconnect() function and then BB8 turns on an "orange" light and reminds like that until I connect to it again.
That is a problem with the battery life. Is it a bug?

Using Sphero v0.9.2

JBX028, 2016-09-03:
Hi,
Any news about this bug? I am face to exactly the same issue....

thanks in advance

JBX028, 2016-09-19:
Hi,
Is there anyone from sphero looking at this repo to answer questions like this one? Everything else works perfectly except the disconnect function however the app on my iphone disconnects properly from my bb8 when I close the app. The firmware is the most recent one because I have been prompted 2 days ago to update it.
This is very frustrating and I am wondering if all developers are face to the same issue because I am surprise to see only this post related to that issue.
Thanks in advance