sphero-inc/sphero.js #80: After a few seconds my BB8 connection is failling On Mac OSX

branstraub, 2016-06-20:
`var sphero = require("sphero"),
    orb = sphero("4aed76c113e84c28be75b49eb45566ea"); // change BLE address accordingly

orb.connect(function() {
  // roll Sphero forward
  orb.roll(150, 0);
  orb.roll(10, 0);
  orb.roll(80, 0);

  
  });
});`

[image]

ShahNewazKhan, 2017-02-22:
I am facing the same issue where OSX is unable to get a response back from BB8 once a command is issued:

```
Unhandled rejection Error: Command sync response was lost.
    at Sphero.handler (/Users/shah/projects/node/ai8/node_modules/sphero/lib/sphero.js:252:21)
From previous event:
    at Sphero.command (/Users/shah/projects/node/ai8/node_modules/sphero/lib/sphero.js:180:10)
    at Sphero.device.setRgbLed (/Users/shah/projects/node/ai8/node_modules/sphero/lib/devices/sphero.js:369:12)
    at Sphero.device.color (/Users/shah/projects/node/ai8/node_modules/sphero/lib/devices/custom.js:168:19)
    at file:///Users/shah/projects/node/ai8/js/orb.js:7:7[0m
```

My current environment is:

```
ProductName:	Mac OS X
ProductVersion:	10.11.6
BuildVersion:	15G31

node -v
v6.9.5

npm -v 
3.10.10

system_profiler SPBluetoothDataType | sed -n "/Apple Bluetooth Software Version\:/,/Manufacturer\:/p" | tr -s "[\n]" "[ ]" | sed "s:.*Address\: ::g" | sed "s: Manufacturer\:.*::g" | grep "[[:graph:]]"

80-E6-50-1B-50-31 Bluetooth Low Energy Supported: Yes Handoff Supported: Yes Instant Hot Spot Supported: Yes
```

YBadiss, 2017-03-04:
I have the same issue, as far as I can tell BB8 is not sending back acks/responses that the framework is expecting. I increased the timeout when initializing the bb8 object but it just waits longer before showing an error.

Here is a very simple ping exhibiting the issue:

```javascript
var sphero = require("sphero"),                                                           
    bb8 = sphero("368062352763460e9591981b2974cac5", {timeout: 2000});                    
                                                                                          
bb8.connect(function() {                                                                  
  bb8.ping(function(err, data) {                                                          
    console.log(err || "data: " + data);                                                  
  });                                                                                     
});

andriitishchenko, 2017-04-12:
Same with SPRK+
Unhandled rejection Error: Command sync response was lost.
> at Sphero.handler (/Users/andrux/Documents/Sphero/node_modules/sphero/lib/sphero.js:252:21)
>     at ontimeout (timers.js:386:14)
>     at tryOnTimeout (timers.js:250:5)
>     at Timer.listOnTimeout (timers.js:214:5)

avidj, 2018-04-06:
same with sprk+, ditto