sphero-inc/sphero.js #81: Unable to connect to multiple devices (Ollie and/or BB-8)

WslF, 2016-10-10:
Hi!

I'm running Ubuntu 16.04 lts and my bluetooth adapter is  BCM20702A0 Bluetooth 4.0. 

It 100% physically could work with few devices because I could control 2 Ollies at the same time (first from official SDK and the second one by unofficial python SDK). But I couldn't control both of them at the same time from the official SDK.

My code:
````
#Test.js
```
````
var sphero = require("sphero");

ollie1 = sphero('XX:XX:XX:XX:XX:XX', {timeout: 1000});
ollie2 = sphero('YY:YY:YY:YY:YY:YY', {timeout: 1000});

ollie1.connect(function() {
	console.log("Ollie1 connected");
	setInterval(function() {
			ollie1.randomColor(function(err, data) {
				console.log("Ollie1 set random color");
			});

	}, 1000);
});

ollie2.connect(function() {
	console.log("Ollie2 connected");
	setInterval(function() {
			ollie2.randomColor(function(err, data) {
				console.log("Ollie2 set random color");
			});

	}, 1000);
});

```
When I execute it only first Ollie changing color.
########################################################
I divided code into two different files and locate them in the different directories:
````
#Test1.js
```
````
var sphero = require("sphero");

ollie1 = sphero('XX:XX:XX:XX:XX:XX', {timeout: 1000});

ollie1.connect(function() {
	console.log("Ollie1 connected");
	setInterval(function() {
			ollie1.randomColor(function(err, data) {
				console.log("Ollie1 set random color");
			});

	}, 1000);
});

```

````
#Test2.js
```
````
var sphero = require("sphero");

ollie2 = sphero('YY:YY:YY:YY:YY:YY', {timeout: 1000});

ollie2.connect(function() {
	console.log("Ollie2 connected");
	setInterval(function() {
			ollie2.randomColor(function(err, data) {
				console.log("Ollie2 set random color");
			});

	}, 1000);
});

```

When I execute second one It trough exceptions:
```
/home/myPath2/node_modules/sphero/lib/adaptors/ble.js:405
  return this._connectedService(serviceId).characteristics;
                                          ^

TypeError: Cannot read property 'characteristics' of null
    at Adaptor._connectedCharacteristics (/home/myPath2/node_modules/sphero/lib/adaptors/ble.js:405:43)
    at Adaptor._connectCharacteristic (/home/myPath2/node_modules/sphero/lib/adaptors/ble.js:358:12)
    at /home/myPath2/node_modules/sphero/lib/adaptors/ble.js:303:12
    at Adaptor._connectService (/home/myPath2/node_modules/sphero/lib/adaptors/ble.js:339:5)
    at /home/myPath2/node_modules/sphero/lib/adaptors/ble.js:301:10
    at Adaptor._connectBLE (/home/myPath2/node_modules/sphero/lib/adaptors/ble.js:325:5)
    at Adaptor.getCharacteristic (/home/myPath2/node_modules/sphero/lib/adaptors/ble.js:300:8)
    at Adaptor.writeServiceCharacteristic (/home/myPath2/node_modules/sphero/lib/adaptors/ble.js:275:8)
    at Adaptor.setTXPower (/home/myPath2/node_modules/sphero/lib/adaptors/ble.js:204:8)
    at /home/myPath2/node_modules/sphero/lib/adaptors/ble.js:157:10
    at /home/myPath2/node_modules/sphero/lib/adaptors/ble.js:276:25
    at /home/myPath2/node_modules/sphero/lib/adaptors/ble.js:302:20
    at /home/myPath2/node_modules/sphero/lib/adaptors/ble.js:348:9
    at Peripheral.<anonymous> (/home/myPath2/node_modules/noble/lib/peripheral.js:73:7)
    at Peripheral.g (events.js:260:16)
    at emitOne (events.js:77:13)
```

Thanks in advance!

jonbgallant, 2016-10-15:
I resolved this by commenting out ble.stopScanning() in /adaptors/ble.js.  Please see this commit for the exact change. https://github.com/jonbgallant/sphero.js/commit/788e944ca793493f3238d65249d7bfb5e9ad72c3

You can either make that change locally or you can use my fork.

scheunemann, 2017-05-25:
# My systems:
- Linux 4.10.13-1-ARCH x86_64 GNU/Linux (Some intel in-built wifi/bluetooth chip)
- Linux  4.9.25-1-ARCH  armv71 GNU/Linux (0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI Mode))

On both: node@v6.10.3, noble@1.8.1

# Needs:
I want to connect to two BB8s.

# Problem:
I have the same results as above.

With one script attempting to connect to two ollies, some sample output might be:

```
# node Test.js
Ollie2 connected
noble warning: unknown peripheral xxxxxxxxxxxx
# node Test.js
Ollie1 connected
# node Test.js
Ollie2 connected
# node Test.js
Ollie1 connected
noble warning: unknown peripheral yyyyyyyyyyyy
# node Test.js
Ollie1 connected
Ollie2 connected
```

The first system doesn't show the warnings.

scheunemann, 2017-06-16:
See pull request here: https://github.com/orbotix/sphero.js/pull/89