sphero-inc/sphero.js #85: BB-8 command sync response error

toonr, 2017-03-06:
Hi,

I'm trying to program BB-8 with the Javascript API. I'm on Windows 10 and have installed all the necessary dependencies, and I am using a bluetooth 4.0 USB. When I run a basic script, f.e. just change the color of BB-8, the robot does it successfully. But when trying to get a response back, I always get the error: 

_Unhandled rejection Error: Command sync response was lost.
    at Sphero.handler (C:\Users\toonr\AppData\Roaming\npm\node_modules\sphero\lib\sphero.js:252:21)
    at ontimeout (timers.js:380:14)
    at tryOnTimeout (timers.js:244:5)
    at Timer.listOnTimeout (timers.js:214:5)_

Or, when using Cylon.js, a similar error:

_Error: Command sync response was lost.
    at Sphero.cb (C:\Users\toonr\AppData\Roaming\npm\node_modules\cylon-sphero-ble\node_modules\sphero\lib\sphero.js:243:21)
    at ontimeout (timers.js:380:14)
    at tryOnTimeout (timers.js:244:5)
    at Timer.listOnTimeout (timers.js:214:5)_

So executing a command (or multiple) is working perfectly, but I can't seem to get data back from the BB-8. Any ideas on what the cause could be? Thanks in advance!

xiangzhi, 2017-03-16:
It seemed to be a issue with the newest version noble(1.8.0). You should be able to avoid this by rolling back to noble@1.7.0. It will be great if we can figure out what in noble broke the code.

toonr, 2017-03-18:
Thanks for the response! I'll test it out this week and will report back here.

scheunemann, 2017-03-28:
I can confirm the experienced behavior with using my BB8.

# Minimal script:

```javascript
var sphero = require("sphero"),
    bb8 = sphero("C9:35:5F:A9:34:79"); // change BLE address accordingly

bb8.connect(function() {
  // roll BB-8 in a random direction, changing direction every second
  setInterval(function() {
    bb8.ping();
  }, 1000);
});
```

# Error message (every second):

> Unhandled rejection Error: Command sync response was lost.
>     at Sphero.handler (<NODE_MODULES>/sphero/lib/sphero.js:252:21)                                                                                              
>     at ontimeout (timers.js:380:14)                                                                                                                                    
>     at tryOnTimeout (timers.js:244:5)                                                                                                                                  
>     at Timer.listOnTimeout (timers.js:214:5)   

# System

Installed node modules: `noble@1.8.0` and `sphero@0.9.2`.

OS: Arch Linux 
Kernel: 4.10.3-1-ARCH #1 SMP PREEMPT Wed Mar 15 09:17:17 CET 2017 x86_64 GNU/Linux
BlueZ: 5.44

# Workaround

When downgrading to `noble@1.7.0` the error does not appear.

closed 2017-04-06 by toonr

jasherone, 2017-11-24:
I think I have very common issue. **Did you manage to fix that?**

I am trying to use this https://github.com/charliegerard/leap_sphero.git

I tested on different versions of MacOS, but I still have the same issue.

I found that the issue could be caused by noble version 1.8.0.
I downgraded noble to 1.7.0, but I still have the issue.

**Current Config**
Mac OSX 10.12.6
Sphero model: S003, SPRK (not plus)
   "express": "^4.16.2",
    "leapjs": "^0.6.4",
    "noble": "^1.7.0",
    "serialport": "^3.1.2",
    "sphero": "^0.9.2"

**Issue**
I only send roll commands to sphero, and I limit them to one command every 200ms.

Fist few commands execute instantly, but then after 10-20 seconds (about 30-50 commands), they suddenly they start lagging up to few seconds long lags and I was getting following message

Unhandled rejection Error: Command sync response was lost.
    at Sphero.handler (/Users/martin/Workspace/LEAP Sphero MJ/node_modules/sphero/lib/sphero.js:252:21)
    at ontimeout (timers.js:475:11)
    at tryOnTimeout (timers.js:310:5)
    at Timer.listOnTimeout (timers.js:270:5)

I tried sending commands more often and less often, it made no difference.

And I also tried waiting (for example 60 seconds) and then started sending commands. It looks like Sphero starts to lag 10-20 seconds after I start to send commands no matter how many commands I send and how much I waited before.

_The error line is in this funcion:_

/**
 * Adds a promise to the queue, to be executed when a response
 * gets back from the sphero.
 *
 * @private
 * @param {Array} cmdPacket the bytes array to be send through the wire
 * @param {Function} resolve function to be triggered on success
 * @param {Function} reject function to be triggered on failure
 * @example
 * sphero._execCommand(packet, resolve, reject);
 * @return {void}
 */
Sphero.prototype._queuePromise = function(cmdPacket, resolve, reject) {
  var seq = cmdPacket[4];
  var handler = function(err, packet) {
    clearTimeout(this.responseQueue[seq].timeoutId);
    this.responseQueue[seq] = null;
    this.busy = false;
    if (typeof resolve === "function") {
      if (!err && !!packet) {
        resolve(packet);
      } else {
      var error = new Error("Command sync response was lost.");
        reject(error);
      }
    }
    this._execCommand();
  };

toonr, 2017-11-24:
Seems like your issue isn't quite similar to mine, since I got the error immediately after connecting to the Sphero and sending a single command. Downgrading the noble package partially solved the problem for me. But I ended up using [Cylon.js](https://cylonjs.com/), which did the trick for me.