sphero-inc/sphero.js #87: Can't connect to BB-8 (advertisement-discovery.js hangs)

ianbrandt, 2017-06-12:
Greetings,

I'm trying to connect to my BB-8, but when I run `advertisement-discovery.js` it just hangs and never prints anything.

I can connect with the iOS app.  The first time it did a firmware update.

I make sure BB-8 is fully charged.  Then I remove him from the charging station, and set him on the floor right next to my MacBook Pro.  BB-8 doesn't activate or light up when I do this.  The I run `advertisement-discovery.js`, but nothing prints.  I tried moving BB-8 all around in close proximity to the laptop. 

I tried [resetting BB-8](https://support.sphero.com/support/solutions/articles/9000064337-how-do-i-reset-bb-8-), which wakes him up, and then repeating the above procedure with him active, but no change.

package.json:
```
[...]
  "dependencies": {
    "noble": "^1.8.1",
    "sphero": "^0.9.2"
  }
}
```

noble and sphero install:
```
$ npm install

> xpc-connection@0.1.4 install /Users/ibrandt/Development/BrandtAcademy/git/bb8/node_modules/xpc-connection
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
../src/XpcConnection.cpp:143:44: warning: 'GetRealNamedProperty' is deprecated [-Wdeprecated-declarations]
      Local<Value> propertyValue = object->GetRealNamedProperty(propertyName->ToString());
                                           ^
/Users/ibrandt/.node-gyp/6.10.3/include/node/v8.h:2923:30: note: 'GetRealNamedProperty' has been explicitly marked
      deprecated here
                Local<Value> GetRealNamedProperty(Local<String> key));
                             ^
1 warning generated.
  SOLINK_MODULE(target) Release/binding.node
clang: warning: libstdc++ is deprecated; move to libc++ with a minimum deployment target of OS X 10.9 [-Wdeprecated]

> usb@1.2.0 install /Users/ibrandt/Development/BrandtAcademy/git/bb8/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ '/usr/local/Cellar/node@6/6.10.3_1/bin/node',
node-pre-gyp verb cli   '/Users/ibrandt/Development/BrandtAcademy/git/bb8/node_modules/usb/node_modules/.bin/node-pre-gyp',
node-pre-gyp verb cli   'install',
node-pre-gyp verb cli   '--fallback-to-build' ]
node-pre-gyp info using node-pre-gyp@0.6.30
node-pre-gyp info using node@6.10.3 | darwin | x64
node-pre-gyp verb command install []
node-pre-gyp info check checked for "/Users/ibrandt/Development/BrandtAcademy/git/bb8/node_modules/usb/src/binding/usb_bindings.node" (not found)
node-pre-gyp http GET https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v48-darwin-x64.tar.gz
node-pre-gyp http 200 https://github.com/tessel/node-usb/releases/download/1.2.0/usb_bindings-v1.2.0-node-v48-darwin-x64.tar.gz
node-pre-gyp info install unpacking usb_bindings.node
node-pre-gyp info tarball done parsing tarball
node-pre-gyp info validate Running test command: '/usr/local/Cellar/node@6/6.10.3_1/bin/node --eval 'require(\'/Users/ibrandt/Development/BrandtAcademy/git/bb8/node_modules/usb/src/binding/usb_bindings.node\')''
[usb] Success: "/Users/ibrandt/Development/BrandtAcademy/git/bb8/node_modules/usb/src/binding/usb_bindings.node" is installed via remote
node-pre-gyp info ok 

> bluetooth-hci-socket@0.5.1 install /Users/ibrandt/Development/BrandtAcademy/git/bb8/node_modules/bluetooth-hci-socket
> node-gyp rebuild

  SOLINK_MODULE(target) Release/binding.node
clang: warning: libstdc++ is deprecated; move to libc++ with a minimum deployment target of OS X 10.9 [-Wdeprecated]
added 295 packages in 12.177s
```

`advertisement-discovery.js` with debug, hangs indefinitely at "noble scanStart":
```
$ DEBUG=* node ./node_modules/noble/examples/advertisement-discovery.js
  noble addressChange 6c:40:08:ba:98:f1 +0ms
  yosemite-bindings sendCBMsg: 1, {
  "kCBMsgArgName": "node-1497250667277",
  "kCBMsgArgOptions": {
    "kCBInitOptionShowPowerAlert": 0
  },
  "kCBMsgArgType": 0
} +3ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 6,
  "kCBMsgArgs": {
    "kCBMsgArgState": 5
  }
} +2ms
  yosemite-bindings state change poweredOn +0ms
  noble stateChange poweredOn +0ms
  yosemite-bindings sendCBMsg: 29, {
  "kCBMsgArgOptions": {},
  "kCBMsgArgUUIDs": []
} +1ms
  noble scanStart +0ms
^C
```

MacBook Pro (Retina, 15-inch, Mid 2014):
```
Apple Bluetooth Software Version:	5.0.4f18
  Hardware, Features, and Settings:
  Name:	Ian's MacBook Pro
  Address:	6c-40-08-ba-98-f1
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	Yes
  Instant Hot Spot Supported:	Yes
  Manufacturer:	Broadcom
  Transport:	USB
  Chipset:	20702B0
  Firmware Version:	v137 c9260
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x05AC
  Product ID:	0x8289
  HCI Version:	4.0 (0x6)
  HCI Revision:	0x242C
  LMP Version:	4.0 (0x6)
  LMP Subversion:	0x4189
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Portable
  Composite Class Of Device:	0x38010C
  Device Class (Major):	0x01
  Device Class (Minor):	0x03
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
```

OS X Version 10.12.5 (16F73)
Xcode Version 8.3.3 (8E3004b)
Node.js v6.10.3

ianbrandt, 2017-06-12:
I tried from my iMac, and it works fine.  Perhaps this is more an issue with nobel or hardware than it is with sphero?

The install looked a bit different this time:
```
$ npm install

> xpc-connection@0.1.4 install /Users/ibrandt/Development/BrandtAcademy/bb8/node_modules/xpc-connection
> node-gyp rebuild

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
../src/XpcConnection.cpp:143:44: warning: 'GetRealNamedProperty' is deprecated [-Wdeprecated-declarations]
      Local<Value> propertyValue = object->GetRealNamedProperty(propertyName->ToString());
                                           ^
/Users/ibrandt/.node-gyp/6.10.2/include/node/v8.h:2923:30: note: 'GetRealNamedProperty' has been explicitly marked deprecated here
                Local<Value> GetRealNamedProperty(Local<String> key));
                             ^
1 warning generated.
  SOLINK_MODULE(target) Release/binding.node
clang: warning: libstdc++ is deprecated; move to libc++ with a minimum deployment target of OS X 10.9 [-Wdeprecated]
bb8@1.0.0 /Users/ibrandt/Development/BrandtAcademy/bb8
├─┬ noble@1.8.1 
│ ├── bplist-parser@0.0.6 
│ ├─┬ debug@2.2.0 
│ │ └── ms@0.7.1 
│ └─┬ xpc-connection@0.1.4 
│   └── nan@2.6.2 
└─┬ sphero@0.9.2 
  ├── bluebird@3.5.0 
  └── browser-serialport@2.1.0 

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@^0.5.1 (node_modules/noble/node_modules/bluetooth-hci-socket):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for bluetooth-hci-socket@0.5.1: wanted {"os":"linux,android,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})
```

iMac (Retina 5K, 27-inch, Late 2014):
```
 Apple Bluetooth Software Version:	5.0.4f18
  Hardware, Features, and Settings:
  Name:	Ian's iMac
  Address:	88-63-df-c7-01-e4
  Bluetooth Low Energy Supported:	Yes
  Handoff Supported:	Yes
  Instant Hot Spot Supported:	Yes
  Manufacturer:	Broadcom
  Transport:	USB
  Chipset:	20702B0
  Firmware Version:	v137 c9262
  Bluetooth Power:	On
  Discoverable:	Off
  Connectable:	Yes
  Auto Seek Pointing:	On
  Remote wake:	On
  Vendor ID:	0x05AC
  Product ID:	0x828D
  HCI Version:	4.0 (0x6)
  HCI Revision:	0x242E
  LMP Version:	4.0 (0x6)
  LMP Subversion:	0x4189
  Device Type (Major):	Computer
  Device Type (Complete):	Mac Desktop
  Composite Class Of Device:	0x380104
  Device Class (Major):	0x01
  Device Class (Minor):	0x01
  Service Class:	0x1C0
  Auto Seek Keyboard:	On
```

Same OS X and Xcode version
Node.js v6.10.2

scheunemann, 2017-06-21:
Hi @ianbrandt I can confirm that I operate BB-8 successfully with `noble@1.8.1` and `sphero@0.9.2` on an Arch linux. There was a similar issue https://github.com/orbotix/sphero.js/issues/90 (didn't see your's). 

You may want to test the following only-noble script and check whether it connects to your sphero (to rule out `noble` or `driver` issues ). Should work assuming BB-8's name is kept as `BB-XXXX`.

```
var noble = require('noble');

noble.on('stateChange', function(state) {
    console.log(state);
    if (state === 'poweredOn') {
        noble.startScanning();
    } else {
    }
});

noble.on('discover', function(peripheral) {

    var localName = peripheral.advertisement.localName;
    console.log("discover ", localName);

    if(localName !== undefined && localName !== "" && localName.substring(0, "BB-".length) === "BB-") {
        console.log("Connecting ", localName);

        peripheral.on('connect', function() {
            console.log(localName, 'connected');
        });

        peripheral.on('disconnect', function() {
            console.log(localName, 'disconnected');
        });

        noble.stopScanning();
        peripheral.connect(function(error) {
            if(error) {
                console.log(error);
                return;
            }
            console.log(localName, 'connected!'); 
            noble.startScanning();
        return;
        });
    }

});
```

deus42, 2018-01-27:
I am facing the same issue -- tried different nodes, but nothing helps. Something with noble as it seems