sphero-inc/sphero.js #90: Almost impossible connection with BB-8

kandohar, 2017-06-19:
Hi,

I'm using this Javascript SDK to control my BB-8 but I can barely not connect my pc to my BB-8.

`var sphero = require("sphero");`
`var bb8 = sphero("e0e55aa81faf");`
`console.log('Application started');`
`bb8.connect(function() {`
`   console.log("Connect");`
`}`

This piece of code will most of the time prompt : `Application started` only.
Perhaps, 1 in 10 time, there is the `Connect` message.

Am I missing somethings ?
Should I press the dock side button ? (BTW what does this button ?)
Is there a precise time when I should start my app ?

The advertisement-discovery.js and peripheral-explorer.js works (most of the time).
The official App works too (most of the time).

(Windows 10, Node v6.10.2, Sphero 0.9.2)

scheunemann, 2017-06-19:
Hi @kandohar ... I am not on Windows. Your script works fine for me (beside the small syntax issue (missing closing bracket  `);` at the end).

Can you try to only use pure `noble` to connect and post the output? The following script may discover all Bluetooth devices and tries to connect to all devices with the prefix "BB-":

```
var noble = require('noble');

noble.on('stateChange', function(state) {
    console.log(state);
    if (state === 'poweredOn') {
        noble.startScanning();
    } else {
    }
});

noble.on('discover', function(peripheral) {

    var localName = peripheral.advertisement.localName;
    console.log("discover ", localName);

    if(localName !== undefined && localName !== "" && localName.substring(0, "BB-".length) === "BB-") {
        console.log("Connecting ", localName);

        peripheral.on('connect', function() {
            console.log(localName, 'connected');
        });

        peripheral.on('disconnect', function() {
            console.log(localName, 'disconnected');
        });

        noble.stopScanning();
        peripheral.connect(function(error) {
            if(error) {
                console.log(error);
                return;
            }
            console.log(localName, 'connected!'); 
            noble.startScanning();
        return;
        });
    }

});
```

Does that work for you?

kandohar, 2017-06-19:
Sorry for the Copy/Paste fail...

I tried your code but it only prints `poweredOn` (I waited 2 minutes).

The BB-8 is on its dock, the dock is pluged on my desktop and I use a bluetooth usb dongle pluged just in front of the dock.

Does the BB-8 need to be in a special state ? Should I press the button ?

scheunemann, 2017-06-19:
Okay, that means there are either no BLE devices in reach, your dongle cannot scan for such devices properly or the BB8 is not advertising messages. In any case, it is not an issue with `sphero.js` (this repository).

---
What happens if you press the button of the dock? Do you see the three blue LED lights flashing/blinking?

kandohar, 2017-06-20:
Ok, yes if I plug the dock and if I put the BB-8 on the dock, the three blue leds are blinking.
If I push the button, BB-8 is flashing orange (quickly) then blue (slowly), then it start moving.

scheunemann, 2017-06-21:
Hi @kandohar considering that the official App works most of the time, I would guess it has more something to do with your BLE stack. I would suggest you try to connect to (or at least scan for) another BLE device and check if everything is fine on that side.

kandohar, 2017-06-21:
Thanks a lot for your helps.
I see, I will check drivers, perhaps another dongle and another devices.

closed 2017-06-21 by kandohar