sphero-inc/sphero.js #93: Sphero SPRK controlled via Mac fills up queue and lags with executing commands

jasherone, 2017-11-25:
Hello there, 
I have a following issue and I hope someone can help me :/

**Quick Description of the issue**
I am trying to use this https://github.com/charliegerard/leap_sphero.git with my Sphero SPRK. But after I start sending commands to Sphero, the queue fills up and Sphero starts to lag in the real world responses to the commands.

**My Question**
1/ Did anyone encounter common issue and managed to fix that?
2/ Any tips on what to study to find out how does the queue or how to debug this?
3/ I was thinking of simply dumping the Sphero queue if it is longer than a few commands, any tips how can I do it please?

**Fixes tried**
1/ I tested on different versions of MacOS (not win and linux however)
2/ I found that the issue could be caused by noble version 1.8.0. I downgraded noble to 1.7.0 as suggested here https://github.com/orbotix/sphero.js/issues/85
3/ Call ping() after connect() as suggested here https://github.com/orbotix/sphero.js/issues/37
4/ @olcar here https://github.com/orbotix/sphero.js/issues/37 said it works for him with Node 5.2.0 so I tried also downgrading to this version

**Current Config**
Mac OSX 10.12.6
Sphero model: S003, SPRK (not plus)
node -v 9.2.0
   "express": "^4.16.2",
    "leapjs": "^0.6.4",
    "noble": "^1.7.0",
    "serialport": "^3.1.2",
    "sphero": "^0.9.2"

**Detailed Description of the Issue**
I only send roll commands to sphero, and I limit them to one command every 200ms.

Fist few commands execute instantly, but then after 10-20 seconds (about 30-50 commands), they suddenly start filling in the queue and execution starts lagging (up to few seconds long lags) and I was getting following message:

Unhandled rejection Error: Command sync response was lost.
    at Sphero.handler (/Users/martin/Workspace/LEAP Sphero MJ/node_modules/sphero/lib/sphero.js:252:21)
    at ontimeout (timers.js:475:11)
    at tryOnTimeout (timers.js:310:5)
    at Timer.listOnTimeout (timers.js:270:5)

I tried sending commands more often and less often, it made no difference.

And I also tried waiting (for example 60 seconds) and then started sending commands. It looks like Sphero starts to lag 10-20 seconds after I start to send commands no matter how many commands I send and how much I waited before.

**_The error line is in this funcion:_**

/**
 * Adds a promise to the queue, to be executed when a response
 * gets back from the sphero.
 *
 * @private
 * @param {Array} cmdPacket the bytes array to be send through the wire
 * @param {Function} resolve function to be triggered on success
 * @param {Function} reject function to be triggered on failure
 * @example
 * sphero._execCommand(packet, resolve, reject);
 * @return {void}
 */
Sphero.prototype._queuePromise = function(cmdPacket, resolve, reject) {
  var seq = cmdPacket[4];
  var handler = function(err, packet) {
    clearTimeout(this.responseQueue[seq].timeoutId);
    this.responseQueue[seq] = null;
    this.busy = false;
    if (typeof resolve === "function") {
      if (!err && !!packet) {
        resolve(packet);
      } else {
      var error = new Error("Command sync response was lost.");
        reject(error);
      }
    }
    this._execCommand();
  };

jasherone, 2017-11-25:
I just noticed that after the queue is full and it starts to throw errors, it empties the queue approximately every half second (I attached the log below). Then after fully emptying the queue it starts to lag immediately.

`
Commands sent:  177  = = = = = = Time Running:  39.431999921798706  = = = = = = = Queue length:  9
Commands sent:  177  = = = = = = Time Running:  39.467000007629395  = = = = = = = Queue length:  9
Commands sent:  177  = = = = = = Time Running:  39.501999855041504  = = = = = = = Queue length:  9
Commands sent:  177  = = = = = = Time Running:  39.53699994087219  = = = = = = = Queue length:  9
Commands sent:  177  = = = = = = Time Running:  39.57099986076355  = = = = = = = Queue length:  9
Commands sent:  177  = = = = = = Time Running:  39.60699987411499  = = = = = = = Queue length:  9
Commands sent:  177  = = = = = = Time Running:  39.64099979400635  = = = = = = = Queue length:  9
Commands sent:  177  = = = = = = Time Running:  39.675999879837036  = = = = = = = Queue length:  9
Commands sent:  177  = = = = = = Time Running:  39.710999965667725  = = = = = = = Queue length:  9
Commands sent:  177  = = = = = = Time Running:  39.74499988555908  = = = = = = = Queue length:  9
Commands sent:  177  = = = = = = Time Running:  39.77999997138977  = = = = = = = Queue length:  9
Commands sent:  177  = = = = = = Time Running:  39.81499981880188  = = = = = = = Queue length:  9
Commands sent:  177  = = = = = = Time Running:  39.848999977111816  = = = = = = = Queue length:  9
Commands sent:  177  = = = = = = Time Running:  39.883999824523926  = = = = = = = Queue length:  9
Unhandled rejection Error: Command sync response was lost.
    at Sphero.handler (/Users/martin/Workspace/LEAP Sphero MJ/node_modules/sphero/lib/sphero.js:257:21)
    at ontimeout (timers.js:475:11)
    at tryOnTimeout (timers.js:310:5)
    at Timer.listOnTimeout (timers.js:270:5)

Commands sent:  177  = = = = = = Time Running:  39.918999910354614  = = = = = = = Queue length:  8
Commands sent:  177  = = = = = = Time Running:  39.9539999961853  = = = = = = = Queue length:  8
Commands sent:  177  = = = = = = Time Running:  39.99000000953674  = = = = = = = Queue length:  8
Commands sent:  177  = = = = = = Time Running:  40.02300000190735  = = = = = = = Queue length:  8
Commands sent:  177  = = = = = = Time Running:  40.05799984931946  = = = = = = = Queue length:  8
Commands sent:  177  = = = = = = Time Running:  40.09299993515015  = = = = = = = Queue length:  8
Commands sent:  177  = = = = = = Time Running:  40.126999855041504  = = = = = = = Queue length:  8
Commands sent:  177  = = = = = = Time Running:  40.16199994087219  = = = = = = = Queue length:  8
Commands sent:  177  = = = = = = Time Running:  40.1969997882843  = = = = = = = Queue length:  8
Commands sent:  177  = = = = = = Time Running:  40.23099994659424  = = = = = = = Queue length:  8
Commands sent:  177  = = = = = = Time Running:  40.26599979400635  = = = = = = = Queue length:  8
Commands sent:  177  = = = = = = Time Running:  40.30299997329712  = = = = = = = Queue length:  8
Commands sent:  177  = = = = = = Time Running:  40.334999799728394  = = = = = = = Queue length:  8
Commands sent:  177  = = = = = = Time Running:  40.370999813079834  = = = = = = = Queue length:  8
Commands sent:  177  = = = = = = Time Running:  40.407999992370605  = = = = = = = Queue length:  8
Unhandled rejection Error: Command sync response was lost.
    at Sphero.handler (/Users/martin/Workspace/LEAP Sphero MJ/node_modules/sphero/lib/sphero.js:257:21)
    at ontimeout (timers.js:475:11)
    at tryOnTimeout (timers.js:310:5)
    at Timer.listOnTimeout (timers.js:270:5)

Commands sent:  177  = = = = = = Time Running:  40.43999981880188  = = = = = = = Queue length:  7
Commands sent:  177  = = = = = = Time Running:  40.47499990463257  = = = = = = = Queue length:  7
Commands sent:  177  = = = = = = Time Running:  40.50999999046326  = = = = = = = Queue length:  7
Commands sent:  177  = = = = = = Time Running:  40.543999910354614  = = = = = = = Queue length:  7
Commands sent:  177  = = = = = = Time Running:  40.5789999961853  = = = = = = = Queue length:  7
Commands sent:  177  = = = = = = Time Running:  40.61399984359741  = = = = = = = Queue length:  7
Commands sent:  177  = = = = = = Time Running:  40.6489999294281  = = = = = = = Queue length:  7
Commands sent:  177  = = = = = = Time Running:  40.68400001525879  = = = = = = = Queue length:  7
Commands sent:  177  = = = = = = Time Running:  40.71799993515015  = = = = = = = Queue length:  7
Commands sent:  177  = = = = = = Time Running:  40.753000020980835  = = = = = = = Queue length:  7
Commands sent:  177  = = = = = = Time Running:  40.787999868392944  = = = = = = = Queue length:  7
Commands sent:  177  = = = = = = Time Running:  40.8219997882843  = = = = = = = Queue length:  7
Commands sent:  177  = = = = = = Time Running:  40.85699987411499  = = = = = = = Queue length:  7
Commands sent:  177  = = = = = = Time Running:  40.89199995994568  = = = = = = = Queue length:  7
Unhandled rejection Error: Command sync response was lost.
    at Sphero.handler (/Users/martin/Workspace/LEAP Sphero MJ/node_modules/sphero/lib/sphero.js:257:21)
    at ontimeout (timers.js:475:11)
    at tryOnTimeout (timers.js:310:5)
    at Timer.listOnTimeout (timers.js:270:5)
`

xocialize, 2018-07-13:
Could it be related to this bit of code in sphero/lib/sphero.js?

/**
 * Adds a sphero command to the queue
 *
 * @private
 * @param {Array} cmdPacket the bytes array to be send through the wire
 * @param {Function} resolve function to be triggered on success
 * @param {Function} reject function to be triggered on failure
 * @example
 * this._queueCommand(cmdPacket, resolve, reject);
 * @return {void}
 */
Sphero.prototype._queueCommand = function(cmdPacket, resolve, reject) {
  if (this.commandQueue.length === 20) { //I lowered this from 256 
    this.commandQueue.shift();
  }

  this.commandQueue.push({ packet: cmdPacket, resolver: resolve, rejecter: reject });
};