sphero-inc/sphero.js #94: Unable to connect to R2D2

robm92, 2017-12-19:
R2D2 is showing up as a BLE device using LightBlue for Mac.
I'm able to use this to obtain the UUID for the sphero sdk connection.

`    var sphero = require('sphero');
    var r2 = sphero('73ECD53D-B7FE-47E1-B86E-9AF39A880ABE');

    r2.connect(function() {
      console.log("Now connected to R2");

      //The Ping command verifies that R2 is awake and receiving commands.
      r2.ping(function(err, data) {
        console.log(err || data);
      });`

When running, the output is simply:

`  noble addressChange 28:f0:76:12:44:91 +0ms
  yosemite-bindings sendCBMsg: 1, {
  "kCBMsgArgName": "node-1513679700187",
  "kCBMsgArgOptions": {
    "kCBInitOptionShowPowerAlert": 0
  },
  "kCBMsgArgType": 0
} +3ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 4,
  "kCBMsgArgs": {
    "kCBMsgArgState": 5
  }
} +2ms`

mamisano, 2017-12-23:
@robm92  I am experiencing the same exact issue. MacOS 10.13.2

`  noble addressChange f4:5c:89:b0:bf:ae +0ms
  yosemite-bindings sendCBMsg: 1, {
  "kCBMsgArgName": "node-1514046739655",
  "kCBMsgArgOptions": {
    "kCBInitOptionShowPowerAlert": 0
  },
  "kCBMsgArgType": 0
} +2ms
  yosemite-bindings xpcEvent: {
  "kCBMsgId": 4,
  "kCBMsgArgs": {
    "kCBMsgArgState": 5
  }
} +1ms`

joseph-holland, 2018-01-03:
Seems this is the same issue as #95 and #93. The newer versions seem to have different characteristics presented than the older Spheros.