sphero-inc/sphero.js #95: Unable to connect to the Sphero Mini

xocialize, 2017-12-27:
My build environment is a raspberry pi 3 with the "stretch" as the OS.  I can successfully connect and control the sprk+ but I receive the following error when trying to connect to the mini.

/var/public/node_modules/sphero/lib/adaptors/ble.js:405
 return this._connectedService(serviceId).characteristics;
 ^

TypeError: Cannot read property 'characteristics' of null
 at Adaptor._connectedCharacteristics (/var/public/node_modules/sphero/lib/adaptors/ble.js:405:43)
 at Adaptor._connectCharacteristic (/var/public/node_modules/sphero/lib/adaptors/ble.js:358:12)
 at /var/public/node_modules/sphero/lib/adaptors/ble.js:303:12
 at /var/public/node_modules/sphero/lib/adaptors/ble.js:346:9
 at Peripheral.<anonymous> (/var/public/node_modules/noble/lib/peripheral.js:73:7)
 at Object.onceWrapper (events.js:254:19)
 at Peripheral.emit (events.js:159:13)
 at Noble.onServicesDiscover (/var/public/node_modules/noble/lib/noble.js:208:16)
 at NobleBindings.emit (events.js:159:13)
 at NobleBindings.onServicesDiscovered (/var/public/node_modules/noble/lib/hci-socket/bindings.js:291:8)
 at Gatt.emit (events.js:159:13)
 at Gatt.<anonymous> (/var/public/node_modules/noble/lib/hci-socket/gatt.js:361:12)
 at Gatt.onAclStreamData (/var/public/node_modules/noble/lib/hci-socket/gatt.js:132:26)
 at AclStream.emit (events.js:164:20)
 at AclStream.push (/var/public/node_modules/noble/lib/hci-socket/acl-stream.js:35:10)
 at NobleBindings.onAclDataPkt (/var/public/node_modules/noble/lib/hci-socket/bindings.js:272:15)

julienR2, 2017-12-28:
Hi ! I'm getting the same error on OSX and using Sphero Mini.
I'm connecting via the uuid.
Any idea where to investigate ?

xocialize, 2017-12-28:
I have both a mini and a sprk+.  I've noticed that the output from 
sudo node ./node_modules/noble/examples/peripheral-explorer.js "{UUUID}" is much shorter from the mini.  But it works and shows output.  My best guess is somewhere in the sphero code they are looking to set values based on information just not being provided by the mini.  I'm not sure if this is something they will need to fix with a firmware update or if it's something that can be fixed in the code.

ThomasProctor, 2017-12-28:
I'm running into the same error with a mini on Ubuntu 14.04. I'm running on a Thinkpad X230 with a Broadcom Corp. BCM20702 Bluetooth 4.0.

Just for troubleshooting sake, how are you identifying the mini? I can't find much on the naming conventions for the Sphero Mini, but guessed that it was the device I found with local name "SM-9D49". Does this match the naming convention you're seeing?

Just fyi, here's the output of peripheral-explorer.js that I see:

peripheral with ID c695822c9d49 found
  Local Name        = SM-9D49
  Service Data      = 
  Service UUIDs     = 00010001574f4f2053706865726f2121

services and characteristics:
1800 (Generic Access)
  2a00 (Device Name)
    properties  read
    value       534d2d39443439 | 'SM-9D49'
  2a01 (Appearance)
    properties  read
    value       0000 | '^@^@'
  2a04 (Peripheral Preferred Connection Parameters)
    properties  read
    value       0600180000001900 | '^F^@^X^@^@^@^Y^@'
1801 (Generic Attribute)
  2a05 (Service Changed)
    properties  indicate
00020001574f4f2053706865726f2121
  00020003574f4f2053706865726f2121
    properties  writeWithoutResponse
  00020002574f4f2053706865726f2121
    properties  write, notify
  00020004574f4f2053706865726f2121
    properties  read
    value       000008000c000202 | '^@^@^H^@^L^@^B^B'
  00020005574f4f2053706865726f2121
    properties  writeWithoutResponse, write
180f (Battery Service)
  2a19 (Battery Level)
    properties  read, notify
    value       64 | 'd'
00010001574f4f2053706865726f2121
  00010002574f4f2053706865726f2121
    properties  writeWithoutResponse, write, notify
  00010003574f4f2053706865726f2121
    properties  writeWithoutResponse, write, notify

xocialize, 2017-12-28:
From what I can tell you're info is similar to what I have.  My output is below and the Local Name seems to be similar to yours.

 peripheral with ID d1cfcbccf40e found
  Local Name        = SM-F40E
  Service Data      = 
  Service UUIDs     = 00010001574f4f2053706865726f2121

services and characteristics:
1800 (Generic Access)
  2a00 (Device Name)
    properties  read
    value       534d2d46343045 | 'SM-F40E'
  2a01 (Appearance)
    properties  read
    value       0000 | '  '
  2a04 (Peripheral Preferred Connection Parameters)
    properties  read
    value       0600180000001900 | '     '
1801 (Generic Attribute)
  2a05 (Service Changed)
    properties  indicate
00020001574f4f2053706865726f2121
  00020003574f4f2053706865726f2121
    properties  writeWithoutResponse
  00020002574f4f2053706865726f2121
    properties  write, notify
  00020004574f4f2053706865726f2121
    properties  read
    value       000008000c000202 | '    '
  00020005574f4f2053706865726f2121
    properties  writeWithoutResponse, write
180f (Battery Service)
  2a19 (Battery Level)
    properties  read, notify
    value       64 | 'd'
00010001574f4f2053706865726f2121
  00010002574f4f2053706865726f2121
    properties  writeWithoutResponse, write, notify
  00010003574f4f2053706865726f2121
    properties  writeWithoutResponse, write, notify

For comparison here's the output from my sprk+ which does connect and is working.  Notice how much more info is being output:

peripheral with ID c904e6b3436a found
  Local Name        = SK-436A
  TX Power Level    = 6
  Manufacturer Data = 3530
  Service Data      = 
  Service UUIDs     = 22bb746f2ba075542d6f726568705327

services and characteristics:
1800 (Generic Access)
  2a00 (Device Name)
    properties  read
    value       534b2d34333641 | 'SK-436A'
  2a01 (Appearance)
    properties  read
    value       8000 | '  '
  2a04 (Peripheral Preferred Connection Parameters)
    properties  read
    value       080010000000af00 | '    / '
1801 (Generic Attribute)
  2a05 (Service Changed)
    properties  read, indicate
    value       05ffff | ''
22bb746f2ba075542d6f726568705327
  22bb746f2ba175542d6f726568705327
    properties  writeWithoutResponse, write
  22bb746f2ba675542d6f726568705327
    properties  notify
22bb746f2bb075542d6f726568705327
  22bb746f2bb175542d6f726568705327
    properties  read, write
    value        | ''
  22bb746f2bb275542d6f726568705327
    properties  write
  22bb746f2bb675542d6f726568705327
    properties  read, writeWithoutResponse, write, notify
    value       df | '_'
  22bb746f2bb775542d6f726568705327
    properties  read, writeWithoutResponse, write
    value        | ''
  22bb746f2bb875542d6f726568705327
    properties  read
    value       3c | '<'
  22bb746f2bb975542d6f726568705327
    properties  read
    value       0f00 | ' '
  22bb746f2bba75542d6f726568705327
    properties  read
    value       af00 | '/ '
  22bb746f2bbd75542d6f726568705327
    properties  writeWithoutResponse, write
  22bb746f2bbe75542d6f726568705327
    properties  read, write
    value       1e | ''
  22bb746f2bbf75542d6f726568705327
    properties  read, writeWithoutResponse, write
    value       00 | ' '
  22bb746f3bba75542d6f726568705327
    properties  read, writeWithoutResponse, write
    value       0100 | ' '
00001016d10211e19b2300025b00a5a5
  00001013d10211e19b2300025b00a5a5
    properties  read, write
    value       01 | ''
  00001017d10211e19b2300025b00a5a5
    properties  write
  00001014d10211e19b2300025b00a5a5
    properties  read, notify
    value        | ''
180a (Device Information)
  2a27 (Hardware Revision String)
    properties  read
    value       410000000000 | 'A     '
  2a25 (Serial Number String)
    properties  read
    value       43393a30343a45363a42333a34333a3641 | 'C9:04:E6:B3:43:6A'
  2a24 (Model Number String)
    properties  read
    value       353000 | '50 '
  2a29 (Manufacturer Name String)
    properties  read
    value       53706865726f | 'Sphero'
  2a26 (Firmware Revision String)
    properties  read
    value       312e3535 | '1.55'

joseph-holland, 2017-12-29:
Having same issue with Sphero R2-D2.  My output from peripheral-explorer.js is below and looks similar to the one above from the Sphero Mini.

```
peripheral with ID fc8689fba85e found
  Local Name        = D2-A85E
  Service Data      =
  Service UUIDs     = 00010001574f4f2053706865726f2121

services and characteristics:
1800 (Generic Access)
  2a00 (Device Name)
    properties  read
    value       44322d41383545 | 'D2-A85E'
  2a01 (Appearance)
    properties  read
    value       0000 | ''
  2a04 (Peripheral Preferred Connection Parameters)
    properties  read
    value       0600280000001900 | '('
1801 (Generic Attribute)
  2a05 (Service Changed)
    properties  indicate
00020001574f4f2053706865726f2121
  00020003574f4f2053706865726f2121
    properties  writeWithoutResponse
  00020002574f4f2053706865726f2121
    properties  write, notify
  00020004574f4f2053706865726f2121
    properties  read
    value       1200010004000202 | ''
  00020005574f4f2053706865726f2121
    properties  writeWithoutResponse, write
180f (Battery Service)
  2a19 (Battery Level)
    properties  read, notify
    value       56 | 'V'
00010001574f4f2053706865726f2121
  00010002574f4f2053706865726f2121
    properties  writeWithoutResponse, write, notify
  00010003574f4f2053706865726f2121
    properties  writeWithoutResponse, write, notify
```

ThomasProctor, 2017-12-29:
In lines 64 - 70 of `node_modules/sphero/lib/adaptors/ble.js`, the UUID's corresponding to specific services in Ollies and BB-8s (and apparently SPRK+) are defined:

`Adaptor.BLEService = "22bb746f2bb075542d6f726568705327";
Adaptor.WakeCharacteristic = "22bb746f2bbf75542d6f726568705327";
Adaptor.TXPowerCharacteristic = "22bb746f2bb275542d6f726568705327";
Adaptor.AntiDosCharacteristic = "22bb746f2bbd75542d6f726568705327";
Adaptor.RobotControlService = "22bb746f2ba075542d6f726568705327";
Adaptor.CommandsCharacteristic = "22bb746f2ba175542d6f726568705327";
Adaptor.ResponseCharacteristic = "22bb746f2ba675542d6f726568705327";
`
These UUIDs can be found in the peripheral explorer output that @xocialize posted for the SPRK+, but they don't appear in the mini or R2-D2. I don't think this is a bug in the mini or R2-D2, as the UUIDs in ble.js include the main service UUID. I suspect that Sphero has updated some of its Bluetooth firmware for newer models, and hasn't updated this software. Given that there haven't been updates to sphero.js for 2 years, we probably shouldn't be too surprised.

Presumably, we could fix this specific problem by putting in the corresponding UUIDs for the newer models. I can't see any way to figure out what any of these services mean. Also, given that there seems to have been a bunch of updates, there probably will be other bugs we'll have to fix.

ThomasProctor, 2017-12-29:
I think it's unlikely that this is going to be fixed any time soon. Sphero seems to have more or less abandoned it. I think that if we want to control our Sphero minis with a fully featured code base, we'll have to do it ourselves or use a different project. 

One of the big developers for this project seems to have moved to [the hybrid group](https://github.com/hybridgroup), who actually responded to a [sphero mini feature request on their go based interface](https://github.com/hybridgroup/gobot/issues/474). That group also seems to have an actively maintained [javascript adapter](https://github.com/hybridgroup/cylon-sphero), but it seems like it is dependent on sphero.js.

xocialize, 2017-12-30:
Exploring @ThomasProctor find I've copied the node_modules/sphero/lib/adaptors/ble.js to node_modules/sphero/lib/adaptors/ble2.js changed the service address to "00010001574f4f2053706865726f2121" and passed in a custom  my initialization: 

global.bb8 = sphero('d1cfcbccf40e',{adaptor:"ble2"});

No crash for characteristic.  but when I try and pass in a command I now get:

Unhandled rejection TypeError: this.connection.write is not a function
    at Sphero._execCommand (/var/public/node_modules/sphero/lib/sphero.js:227:21)
    at /var/public/node_modules/sphero/lib/sphero.js:187:10
    at Promise._execute (/var/public/node_modules/bluebird/js/release/debuggability.js:303:9)
    at Promise._resolveFromExecutor (/var/public/node_modules/bluebird/js/release/promise.js:483:18)
    at new Promise (/var/public/node_modules/bluebird/js/release/promise.js:79:10)
    at Sphero.command (/var/public/node_modules/sphero/lib/sphero.js:183:10)
    at Sphero.device.roll (/var/public/node_modules/sphero/lib/devices/sphero.js:445:12)
    at app.post (/var/public/app/routes/note_routes.js:43:14)
    at Layer.handle [as handle_request] (/var/public/node_modules/express/lib/router/layer.js:95:5)
    at next (/var/public/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/var/public/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/var/public/node_modules/express/lib/router/layer.js:95:5)
    at /var/public/node_modules/express/lib/router/index.js:281:22
    at Function.process_params (/var/public/node_modules/express/lib/router/index.js:335:12)
    at next (/var/public/node_modules/express/lib/router/index.js:275:10)
    at expressInit (/var/public/node_modules/express/lib/middleware/init.js:40:5)

ThomasProctor, 2017-12-30:
Hmm... I tried something similar, but didn't get that error. What did you set as the service address?

Within ble.js, `Adaptor.BLEService` is not the service address of your sprk+ - `Adaptor.RobotControlService` is. I set `Adaptor.RobotControlService` to the service address and got the same error that we did before. I had assumed that that was because `Adaptor.BLEService` was still wrong, but maybe that error is not raised because you've figured out the right thing.

I was thinking of just trying every permutations of the UUIDs that our minis show, but I think that I'd prefer to use gobot instead.

xocialize, 2017-12-31:
I used: 00010001574f4f2053706865726f2121 from the output from the service UUID out of peripheral-explorer.js

igbopie, 2018-02-26:
I am trying to build a js lib for the new toys, I will get back when I got it working :)

igbopie, 2018-03-13:
@xocialize Got it working with my sdk! adding more functionality right now and making it more robust implementation.

xocialize, 2018-03-13:
@igbopie  That's very impressive.  Let me know when you're at a point for some testing.  I've run into a few issues with disconnecting the spheros with BLE that I'd love to discuss when you're at that point if you're willing.

igbopie, 2018-03-15:
@xocialize https://youtu.be/nwYBTmtn5xY
Soon!

igbopie, 2018-03-22:
@xocialize I think it is ready for some testing, I can't figure out the sensor data, I probably need a hand over there :)

MProx, 2018-05-18:
Hi all, I have recently got a Sphero Mini and have decided to try to put together a Python library for it (because I mainly code in Python). So far, I have managed to establish a BLE connection using Bluepy, determined the available services and characteristics, and then - by reverse-engineering @igbopie 's javascript library - wake up the device, and (so far) change the LED colour. But @igbopie , I have a few questions for you.

How did you manage to determine the correct sequence of actions when writing your library? For example, in your start() function in core.js, you begin by writing the string 'usetheforce...band' to the "antiDoSCharacteristic" characteristic. This seems to stop my device from going to sleep after 10 seconds, but how did you know to write that phrase to that characteristic? Did you analyze the hcidump when coding using the Sphero Edu app? If so, it must have taken ages. Similarly, in the method to change the colour of an LED based on a RGB values (setMainLedColor in user-io.js), the first two bytes of the payload are 0 and 144, with the R, G and B byte vales thereafter. How did you determine these and other similar values?

Your code has been very helpful for me (thank you!) but I'm struggling to follow because Javascript is something that I have almost no experience with.

staceylevine, 2018-09-05:
@igbopie I am having the same issue of not detecting my Sphero mini. I'm actually using your library and it simply just says scanning devices and never finds anything.   Suggestgions are appreciated.