sphero-inc/sphero.js #96: Support for BB-9E

Ezor, 2017-12-29:
Hi,

I can't connect to my BB-9E using the API. It looks like the Characteristics UUID have changed.

```
/Users/ezor/Projets/Node/bb9/node_modules/sphero/lib/adaptors/ble.js:405
  return this._connectedService(serviceId).characteristics;
                                          ^

TypeError: Cannot read property 'characteristics' of null
    at Adaptor._connectedCharacteristics (/Users/ezor/Projets/Node/bb9/node_modules/sphero/lib/adaptors/ble.js:405:43)
```

Can you provide at least the new characteristics to use ?

```
Adaptor.BLEService = "22bb746f2bb075542d6f726568705327";
Adaptor.WakeCharacteristic = "22bb746f2bbf75542d6f726568705327";
Adaptor.TXPowerCharacteristic = "22bb746f2bb275542d6f726568705327";
Adaptor.AntiDosCharacteristic = "22bb746f2bbd75542d6f726568705327";
Adaptor.RobotControlService = "22bb746f2ba075542d6f726568705327";
Adaptor.CommandsCharacteristic = "22bb746f2ba175542d6f726568705327";
Adaptor.ResponseCharacteristic = "22bb746f2ba675542d6f726568705327";
```

ThomasProctor, 2018-01-03:
I'm guessing this is the same issue as #95. If you're willing to do a bunch of guessing and checking, You can view all the available characteristics - there aren't that many - and guess and check until you get things working right. We talk about how to find the available (unlabeled) characteristics at #95, and there is comparison output for a robot with the proper characteristics.

Ezor, 2018-01-03:
Yes this is probably the same issue.

I used the Light Blue App to discover the services of the BB9 and tried to update the code but no luck. 
I’m afraid this is not just simple as using the good characteristics UUID.

It doesn’t make any sense for Sphero to not updating SDK as they are forced to update their own app (edu/Star Wars) to support new robot.

I’m convinced that it represent no more than few hours of work to update the JavaScript code to support the new robot as they already have all the information from their apps.

joseph-holland, 2018-01-03:
Yeah this is all the same issue really. It seems that the newer firmwares on the Sphero Mini, BB-9E, R2-D2 and R2-Q5 don't reveal the same characteristics as the older robots.

I've tried to discover some of them by logging the Bluetooth traffic on my phone and then examining the log in Wireshark (https://payatu.com/reversing-exploiting-ble-4-0-communication/). Using this and looking at the output from peripheral-explorer.js I was able to get about three of the newer uuids for my R2-D2, but these are no good anyway for the JavaScript SDK. 😤

robm92, 2018-01-05:
@joseph-holland If you have an iPad to hand, download the swift playgrounds app and share the R2D2 project to yourself. You can view the raw swift source in Xcode which contains all the bits you need to connect to all V2 Sphero toys.

joseph-holland, 2018-01-05:
Thanks @robm92. I saw that the SDK on Mac has the details for the new droids. I had hoped that the current JavaScript SDK had supported them and was planning to integrate them into my home automation system, but I actually ended up sending both droids back. It seems they're not as open as I had hoped.

Ezor, 2018-01-05:
I checked out the code of the R2D2 Playgrounds and I think It will be not possible to make the current code work by just using the characteristics of the new robots.
For example there is no more `wake` characteristic, wake command is sent directly to the command characteristics.

michaelh99, 2018-01-13:
Based on this and other issues it sounds like Sphero has washed their hands of the SDK and aren't providing any support?

igbopie, 2018-02-21:
Hi @Ezor, 

I imported the R2D2 playgrounds, modified the prefix the of the droid to support Sphero Mini, (SM-). Then, share it back to the iPad and it worked! I think I can build a JS SDK out of that. I will keep you posted.

Ezor, 2018-02-27:
Thanks @igbopie

igbopie, 2018-02-27:
Hi @Ezor,
Got to wake up my Sphero mini using JS. So excited! :)

igbopie, 2018-03-15:
@Ezor Progress so far: https://youtu.be/nwYBTmtn5xY

Ezor, 2018-03-15:
Amazing !!

DanielSSilva, 2018-04-02:
Any updates regarding the "original" issue. I got a bb9 on christmas as I was hopping I could program it without being limited to do it on mobile (with sphero edu app). Since sphero edu app uses JS, why is it so hard to get it working with other devices ( in my case, RPI3). I've based on other tutorials for BB8, hopping it was the same, but noble is giving me the same error as @Ezor (TypeError: Cannot read property 'characteristics' of null).
I'm kind of disappointed with sphero...

Ezor, 2018-04-03:
I contacted directly the Sphero Support about this problem and here is the answer : 

> Greetings,
> 
> Thanks for your interest in Sphero! BB-9E is a licensed product of Disney so it's app compatibility is limited. So currently there are no plans to expand that, but with that being said we are always working on creating new products as well as finding innovative ways to improve the products we currently have, stay tuned!
> 
> Thank you,

igbopie, 2018-04-03:
Hi @DanielSSilva ,

Sphero uses a new protocol to talk to new toys like BB9E, Sphero-Mini and so on. So far, sphero has not updated any of their SDKs with the new toys. I am developing an addon so we can program those new Toys but I don't have a BB-9E (only Sphero mini, and Lighting McQueen), but, it should work similar to Mini. We can get collaborate to make it work if you are interested.

By the way, the difference between Edu and this is that Edu actually sends the program to the Toy, and the Toy executes it. My SDK sends commands through bluetooth, but the program runs on the computer. 

It is kind of hard to develop this since there is docs and it is a work of trying and guessing. If Sphero could provide some Docs, I could make it work.

Thanks!
https://github.com/igbopie/spherov2.js

igbopie, 2018-04-03:
Ahh I see, but also I think there is an API to add some kind of macros to the sphero (I saw that in the old docs). Some time the old protocol helps to undertand the new one (some payloads are the same).

So, one thing I need is the bluetooth code. For example, lighting mcqueen starts with ‘LM-xxxx’, sphero mini with ‘SM-xxxx’, so bb9 maybe something like ‘B9-xxxx’?.

DanielSSilva, 2018-04-03:
@igbopie funny thing, it's not b9 🤣it's something like 'GF-xxxx' as far as I recall. Later today I'll give you the right name

DanielSSilva, 2018-04-03:
Maybe I'll install Wireshark and try to sniff what's going on between edu app and Sphero (if it's even possible since it's sent by Bluetooth and i don't know if Wireshark catch that). It's a long shot but I might have some luck

igbopie, 2018-04-03:
Here is how you do it:

http://www.fte.com/WebHelp/BPA600/Content/Documentation/WhitePapers/BPA600/Encryption/GettingAndroidLinkKey/RetrievingHCIlog.htm

Then you use wireshark to read that file. 

Maybe we can find another communication channel? Haha

DanielSSilva, 2018-04-03:
@igbopie Thank you for that link, i'll check that when I can.
In the meantime, I'll cross post this issue with [this one](https://github.com/dimorphic/bb8/issues/4) (which later creates issue number 5) where the project owner shares some (hopefully useful) insights. Maybe we can jump into that issue (the number 5) and talk over there. If he is able to collaborate, we can setup some channel that the 3 of us have.

igbopie, 2018-04-03:
I have created a ticket on my project so we can discuss there. 

https://github.com/igbopie/spherov2.js/issues/12

dimorphic, 2018-04-06:
@DanielSSilva @igbopie: Hey guys, thanks for reaching out first of all! Hell yea, I'm up for collaborating, but my time is a bit short atm, i'll try to help as much as i can -.-

I was also thinking in patching / rewriting (+TS) this lib when discussing back with @DanielSSilva , as it's 'outdated' and projects depends on it, including my `bb8` (xbox/myo). Pretty sad that Disney does this with all SW 'branded' Sphero toys, but they don't mind 'borrowing' ideas (Force band); had the same problem with BB8 as it was not supported by `sphero.js` on launch. Anyways...

I agree, let's move the discussion to the new lib : https://github.com/igbopie/spherov2.js/issues/12

Great job @igbopie , you the man! 🍻