sphero-inc/sphero.js #97: Is this project dead?

shannonmoeller, 2018-01-16:
Many open issues. No recent commits. Is this project dead?

annegatchell, 2018-01-16:
I'm really sorry about the lack of activity. We haven't had cycles to allocate to the care of these repos for quite a while, but I hope that it will get picked back up.

shannonmoeller, 2018-01-17:
I understand. Just excited by the idea of programming my R2 with Node.js and hoping for movement on stuff like #94.

joseph-holland, 2018-01-18:
@shannonmoeller It certainly seems so. Looks like the main contributer has moved to another group and there's been no real activity in about two years. 

I actually picked up both R2-D2 and BB-8 droids and ended up returning them as I couldn't program them other than the mobile apps or SDKs on OSX, etc. I had hoped to use the JavaScript SDK to integrate them with my home automation system and had raised a support case to ask about an update to the SDK. I got no response from them and then just gave up and returned the droids.

deus42, 2018-01-28:
I want also to play with bb-8, but facing tons of issues with communicating using BLE on MacOS High Sierra. I was trying a different version of Nodejs, found a fixed noble package which works Node v8. But still cannot run a simple example using sphero.js library.

igbopie, 2018-03-05:
Hey guys, just an FYI, I am working on a node sdk for the new toys. I have been able to talk with the sphero mini and lighting mcqueen. BB8 and others should be similar. Will keep you posted!

igbopie, 2018-03-15:
Progress so far: https://youtu.be/nwYBTmtn5xY

igbopie, 2018-04-18:
@wyseguyonline You can check https://github.com/igbopie/spherov2.js We can add support to R2D2 :)

MProx, 2018-05-16:
@igbopie, it would be incredibly useful if you could put together some kind of reference of the information you have on the Sphero Mini bluetooth communications. I'm trying to create a Python library, because I am not good with Javascript - so reverse-engineering what is out there is not easy for me. I'm finding it very difficult without knowing something about the Sphero Mini's BLE communications protocols (UUIDs, services, characteristics, packet structure). Would you consider creating something like this? (or do you know where I can find it if it exists?)

mneira10, 2018-12-28:
Hi! I was thrilled to get an R2D2 for Christmas and have been trying like crazy to connect to it with Node.js using the Sphero.js package to no avail. Has anyone managed to successfully connect? @igbopie have you managed to add support for R2? 

It really is a shame the repos haven't been updated in over 3 years...

igbopie, 2019-01-02:
I think it works, but since I don't have one, I can't verify