sphero-inc/sphero.js #101: Unhandled rejection TypeError: serialport.SerialPort is not a constructor

moarliman, 2018-04-19:
Ubuntu 17.10: Sphero 2 is connected on serialport rfcomm0. Getting the following error:
```
Unhandled rejection TypeError: serialport.SerialPort is not a constructor
    at Adaptor.open (/home/moarli/Dokumente/sphero/node_modules/sphero/lib/adaptors/serialport.js:55:32)
    at /home/moarli/Dokumente/sphero/node_modules/sphero/lib/sphero.js:130:16
    at Promise._execute (/home/moarli/Dokumente/sphero/node_modules/bluebird/js/release/debuggability.js:303:9)
    at Promise._resolveFromExecutor (/home/moarli/Dokumente/sphero/node_modules/bluebird/js/release/promise.js:483:1
8)
    at new Promise (/home/moarli/Dokumente/sphero/node_modules/bluebird/js/release/promise.js:79:10)
    at Sphero.connect (/home/moarli/Dokumente/sphero/node_modules/sphero/lib/sphero.js:92:10)

    at Object.<anonymous> (/home/moarli/Dokumente/sphero/index.js:4:9)
    at Module._compile (module.js:643:30)
    at Object.Module._extensions..js (module.js:654:10)
    at Module.load (module.js:556:32)
    at tryModuleLoad (module.js:499:12)
    at Function.Module._load (module.js:491:3)
    at Function.Module.runMain (module.js:684:10)
    at startup (bootstrap_node.js:187:16)
    at bootstrap_node.js:608:3
```

inear, 2018-04-21:
Looks like the call to node-serialport in the adaptor is wrong (in sphero/lib/adaptors/serialport.js). It says "new serialport.SerialPort(this.conn, {});" but should be "new serialport(this.conn, {});" Or preferable rename local var "serialport" to "SerialPort" like 
var SerialPort = require("serialport");

Maybe it has to do with this commit: https://github.com/node-serialport/node-serialport/commit/9a98ede77f42820316b6e629e722b165f3d1e40a#diff-4897cea01b861088accb90c155bb3db8L171. That change (back in 2016) hade a depreciation warning but that was probably removed later.

inear, 2018-04-21:
And then I found #99