sphero-inc/sphero.js #102: Details about Sphero Mini communications?

MProx, 2018-05-16:
Would it be possible to release some information about the Sphero Mini's BLE communication protocol? I am trying to create a Python library for this device, but it is nearly impossible without knowing something about the bluetooth services, characteristics, attributes, and how they work.

Just a little information would undoubtedly allow hackers in the community to create their own APIs, and I think that could significantly increase the reach of the Sphero Mini project.

MProx, 2018-07-19:
Lol, one of those is my library :) I posted this request when I was just starting to write it, and I made good progress with advice from people here and elsewhere on the internet (and with a lot of bluetooth packet sniffing). I have been a little busy recently, but I'm hoping to do some more work on it soon. Details about the API would be extremely helpful because the communications protocol behaves very oddly in some ways. 

I did reach out to Sphero's tech support and was essentially told that no API is currently available, but they are hoping to release one eventually. I would be happily surprised if they are actually putting any resources towards this, but I am doubtful.

siegmund42, 2018-07-19:
Sorry, didn't even recognize that :laughing: 
Offtopic: You did a good job with your library so far, the only thing I'm missing at the moment is the possibility of sensor data streaming.

MProx, 2018-07-19:
No worries :) It makes me very happy to see people using it. Sensor data is exactly what I was talking about when I mentioned the odd behavior. I agree, it would be incredibly awesome to crack that particular problem.

MProx, 2018-09-19:
@siegmund42 Hi there. You might be interested to know that I have added some basic sensor functionality to my library. It is very experimental, and crashes are frequent, but you're welcome to give it a go if you'd like.