sphero-inc/sphero.js #103: Not working with Sphero Bolt

mdinoulis, 2019-01-27:
I'm trying to connect with the Sphero Bolt but when I try and run the example program I get the following error:

/Volumes/Promise Pegasus/Miscellaneous/Programming/Sphero/sphero.js/lib/adaptors/ble.js:45
  var uuid = peripheralId.split(":").join("").toLowerCase();
  
It looks like it's something to do with the Bluetooth connection, however I have installed installed the NodeJS serialport module and sphero serialport as documented.

I did have some problems pairing via Bluetooth though. I can't seem to see or connect to the Bolt through the Bluetooth configuration panel but when I connect through SpheroEdu I can then see the device connected fine. I still can't see anything in /dev though. Although the Sphero documentation does say that the Bolt uses Bluetooth Classic.

Are you able to help?

Cheers,
Dino

mdinoulis, 2019-01-27:
I've just contacted Sphero support and they have told me that the Bolt connects using Bluetooth LE which is why Sphero.js doesn't work with it.

Are there any plans to provide support for the Sphero Bolt anytime soon? Or are there any workarounds that would get it running with the existing code?

Best Regards,
Dino

ogiermaitre, 2019-02-22:
Hello,

I've got the same question. But I'm unclear on why this is different from the BB-8 as in  [this part of the documentation](https://github.com/orbotix/sphero.js#installation-for-bb-8--ollie).

But I'm not able to discover my Bolt using noble.

Regards,
Ogier